esphome/feature-requests #1548: Add support for any binary output or switch to `slow_pwm`

nagyrobi, 2021-12-13:
**Describe the problem you have/What new integration you would like**
`slow_pwm` has `pin` parameter as required, it controls the GPIO directly.
The author perhaps thought that only local pins would be usable. but best would be to support switches directy
as the whole PWM pinciple here is slow enough anyway (minutes, seconds) and uses just ON/OFF, could handle any kind of swithces. 
One could use any kind of configured switch, regardless of platform.

**Please describe your use case for this integration and alternatives you've tried:**
Having a heater (heatpump) which can only be controlled with Modbus protocol. Doesn't have a switch-relay input.
Would like to be able to use a PID climate with slow_pwm to control it. A [coil-based Modbus switch](https://esphome.io/components/switch/modbus_controller.html) is already configured on the node and works as expected. `slow_pwm` could use it...

**Additional context**
Could also be useful for https://github.com/esphome/feature-requests/issues/998.

nagyrobi, 2021-12-13:
@nickw444 @OttoWinter

martgras, 2021-12-15:
Can you please try 
```
external_components:
  - source: github://martgras/esphome@slow-pwm
    components: ["slow_pwm"]

switch:
  - platform: restart
    id: reboot_switch
output:
  - platform: template
    id: outputsplit
    type: binary
    write_action: 
      - then:
         - lambda: return;

  - platform: slow_pwm
#    pin: 5
#    binary_output: outputsplit
#    float_output: <a float type output:>
    switch: reboot_switch
    id: my_slow_pwm
    period: 15s
```

martgras, 2021-12-15:
You can still test the version from github://martgras/esphome@slow-pwm.
However I think it makes more sense to 2 use actions.
I will add a `turn_on` and `turn_off` action instead to make it more flexible

nagyrobi, 2021-12-15:
Hm. Looks like my ESP32 bricked after adding to the config:

```
external_components:
  - source: github://martgras/esphome@slow-pwm
    components: ["slow_pwm"]

output:
- platform: slow_pwm
#  pin: 5
#  binary_output: output1
#  float_output: <a float type output:>
  switch: circuit_01
  id: my_slow_pwm
  period: 15s

switch:
- platform: modbus_controller
  modbus_controller_id: relay24
  id: circuit_01
  name: 01. ${coil_name_suff}
  register_type: coil
  address: 0
  bitmask: 1

```

nagyrobi, 2021-12-15:
Error is:
[17:00:24]ets Jun  8 2016 00:22:57
```
[17:00:24]
[17:00:24]rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
[17:00:24]configsip: 0, SPIWP:0xee
[17:00:24]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:00:24]mode:DIO, clock div:2
[17:00:24]load:0x3fff0018,len:4
[17:00:24]load:0x3fff001c,len:1044
[17:00:24]load:0x40078000,len:8896
[17:00:24]load:0x40080400,len:5828
[17:00:24]entry 0x400806ac
[17:00:25][E][esp32-hal-misc.c:111] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
[17:00:25]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[17:00:25]Core 1 register dump:
[17:00:25]PC      : 0x40175179  PS      : 0x00060130  A0      : 0x800de578  A1      : 0x3ffb1da0  
[17:00:25]A2      : 0x3ffb8ddc  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000001  
[17:00:25]A6      : 0x00060120  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
[17:00:25]A10     : 0x00000000  A11     : 0x00000003  A12     : 0x00000050  A13     : 0x3ffb3c78  
[17:00:25]A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
[17:00:25]EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  
[17:00:25]
[17:00:25]ELF file SHA256: 0000000000000000
[17:00:25]
[17:00:25]Backtrace: 0x40175179:0x3ffb1da0 0x400de575:0x3ffb1dc0 0x401754ad:0x3ffb1de0 0x40175579:0x3ffb1e00 0x400e2f85:0x3ffb1e20 0x400e6d49:0x3ffb1e70 0x400f3042:0x3ffb1fb0 0x4008a1e2:0x3ffb1fd0
[17:00:25]
[17:00:25]Rebooting...
```

martgras, 2021-12-16:
Ok here is my new approach at this -  a lot more flexible 

You have to define `` turn_on_action`` and ``turn_off_action``.    
This examples turns on/off  ``output1`` 

```yaml
external_components:
  - source: github://martgras/esphome@slow-pwm-action
    components: ["slow_pwm"]

esphome:
  name: testing
  on_boot:
    ## set slow_pwm to 25% at startup
    priority: -100
    then:
      output.set_level:
        id: my_slow_pwm
        level: 25%
          

switch:
  - platform: restart
    id: reboot_switch
output:
  - platform: template
    id: output1
    type: binary
    write_action:
      - then:
          - lambda: ESP_LOGD("SLOW PWM","state %d",state);

  - platform: slow_pwm
    # pin: 5
    turn_on_action:
#      - output.turn_on: output1
#   lambda can be used as well
      - lambda: |-
          auto *out = id(output1);
          out->turn_on();
    turn_off_action:
      - output.turn_off: output1
```

nagyrobi, 2021-12-16:
Excellent! Works with switch:

```
- platform: slow_pwm
  id: my_slow_pwm
  period: 10s
  turn_on_action:
    - switch.turn_on: circuit_01
  turn_off_action:
    - switch.turn_off: circuit_01
```

closed 2022-01-09 by oxan