esphome/feature-requests #1549: Allow using lambda in filters

CyanoFresh, 2021-12-13:
**Describe the problem you have/What new integration you would like**

I want to use template input value inside `delayed_off:` filter. Like we can do the same thing with `delay:`

**Please describe your use case for this integration and alternatives you've tried:**

Set the time the light bulb should be on from HA (e.g. 10 seconds or 40 seconds) for motion sensor (motion-activated light).

**Additional context**

albarlow, 2022-01-25:
I would also find this really helpful, I would like to create a 'Motion' sensor that resets immediately as the hardware sensor resets (maybe with a bit of smoothing) and an 'Occupancy' sensor that would be front-end configurable.

At present, I have to reflash the whole device to change the delay, which doesn't allow for much tweaking and my only other options are far more convoluted!

chrishoage, 2022-07-06:
I just found that this isn't possible, here is what I would like to do 

```yaml
binary_sensor:
  - platform: gpio
    pin: GPIO5
    name: "$friendly_name Motion"
    device_class: motion
    filters:
      - delayed_off: !lambda "return id(pir_latency).state"

number:
  - platform: template
    id: pir_latency
    entity_category: config
    name: "$friendly_name PIR Latency"
    unit_of_measurement: ms
    step: 10
    min_value: 0
    max_value: 65000
    optimistic: true
```

This would allow me to configure the delay on the PIR sensor turning off from home assistant, allowing me to use the same package and includes that I use to reduce repetition in the yaml