esphome/feature-requests #1550: Siren entity

Naesstrom, 2021-12-14:
**Describe the problem you have/What new integration you would like**

I've made a siren that uses a GPIO pin to activate a relay and a cheap 12V 120db Siren and a red strobe. Home-Assistant added a siren entity a while back and it would be great to be able to set that in ESPHome
[image]

**Please describe your use case for this integration and alternatives you've tried:**

By setting a variable or something in ESPHome to get Home-Assistant to recognize it as a siren entity

**Additional context**

Link to home-assistant docs: https://developers.home-assistant.io/docs/core/entity/siren

Wferr, 2022-04-21:
This would be helpful!

Naesstrom, 2022-04-23:
> This would be helpful!

I just used the new helper function to change the switch to an alarm in home-assistant. Seems to be working
[image]

closed 2022-07-06 by nagyrobi