esphome/feature-requests #1551: Shutdown "switch" is not supported by the new button component

lanrat, 2021-12-15:
### The problem

With the release of ESPHome `2021.12.0` the new [Button Component](https://esphome.io/components/button/index.html) was added, which can now be used for things like the [Restart Button](https://esphome.io/components/button/restart.html), however it does not yet work with the [Shutdown Switch](https://esphome.io/components/switch/shutdown.html).

Can a Shutdown Button component be added?

Since shutting down, like restarting, does not track an on/off state, it would make sense to make a Shutdown Button component as well.

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12.0

### What platform are you using?

ESP32

### Board

NA

### Component causing the issue

Shutdown Switch

oxan, 2021-12-15:
but maybe I should've transferred it to be a feature request instead :)

jsuanet, 2021-12-15:
I just created PR https://github.com/esphome/esphome/pull/2918 for this.

jesserockz, 2021-12-22:
Closed by esphome/esphome#2918

closed 2021-12-22 by jesserockz