esphome/feature-requests #1554: Get current timestamp in milliseconds

tefracky, 2021-12-17:
**Describe the problem you have/What new integration you would like**

Is ist somehow possible to get the current System time of the ESP in milliseconds? It is possible to get the time in seconds by 

```
time:
  - platform: sntp
    id: sntp_time

...

id(sntp_time).now().timestamp
```

How can I get the timestamp in milliseconds?

Thanks  a lot!

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

bojanpotocnik, 2022-04-15:
I'm trying to get logger component to print timestamps with grater precision than seconds, and stumbled upon this request. Any news on this topic?

nagyrobi, 2022-07-07:
For fast answers, please use the Discord channel.

closed 2022-07-07 by nagyrobi