esphome/feature-requests #1555: Allow disabling display refresh during setup() for display component

Plawasan, 2021-12-19:
**Describe the problem you have/What new integration you would like**
Add an option to the display component to disable screen refresh during setup()

**Please describe your use case for this integration and alternatives you've tried:**
I am using ESPHome with LilyGo T5 e-paper to display weather/climate data. I am using deep_sleep to conserve energy and retrieve data from home assistant. The cycle runs as follows:

1, boot/wakeup
2, display is automatically cleared (this is what I'm trying to suppress) - the screen is now blank
3, wifi connects
4, home assistant data arrives when entities are updated (on a default 60s interval)
5, last entity on_value triggers update of the display component
6, display is refreshed with lambda - the screen now shows what it should
7, esp32 enters deep sleep

In principle this works however the display shows a blank screen between step 2 and 6 which can be up to 60s. If I had the option to suppress the redrawing of the screen on boot/wakeup there would be no period when the display is blank except for when it's being intentionally updated.

**Additional context**
current config:
```
display:
  - platform: t547
    id: t5_display
    rotation: 270
    update_interval: never
    lambda: ...
```

Last entity:
```
  - platform: homeassistant
    entity_id: sensor.openweathermap_forecast_condition
    id: fc_weather
    internal: true
    on_value:
      then:
        - component.update: t5_display
```

ChrisOboe, 2022-01-06:
@Plawasan i'm not a esphome dev, but have you already tried setting auto_clear_enabled attribute at display to false?

Gesiima, 2024-03-10:
It would be nice to have a solution for that request.
I have same issue after wakeup from deepsleep there is a "initialising" process running, that will clear, so if this can be suppressed e-ink display will flicker one time less.
Now my sollution is to disable deepsleep :(