esphome/feature-requests #1557: Winsen - Laser Dust Module - ZH03B

djansen1987, 2021-12-20:
**Describe the problem you have/What new integration you would like**

Please add support for the Winsen ZH03B module. Or if possible in a couple of steps how to add support my self

**Please describe your use case for this integration and alternatives you've tried:**

Measure PM1 pm2.5 pm10 via ESP32 via uart. Tried to add it as an PM1006 but had no luck.

**Additional context**

Source code of @ShaggyDog18 who created an .ino file to make it work on ESP32
https://github.com/ShaggyDog18/SD_ZH03B/blob/master/example/ZH03Bsensor/ZH03Bsensor.ino
Module Manual: [https://www.winsen-sensor.com/d/files/ZH03B.pdf](https://www.winsen-sensor.com/d/files/ZH03B.pdf)

spattinson, 2024-01-24:
REplying to an old request, still relevant I guess. All Winsen sensors seem to operate the same way, maybe there could be a universal winsen integration? they work on uart and I guess you can probably hang a few of them from the same uart, they start up in auto mode where they send every second, they can be switched to "question" mode where they are polled, so if there is more that one on the uart then this provides a way to address each one in turn without them transmitting at the same time and corrupting output.

I've just bought a couple of ZH03B pm sensor,  and I already have some ZE16B CO sensor which I got working through custom include. I'll try setting up both sensors on one uart and see how I get on. Basically just reuse existing code and change the Q&A format

djansen1987, 2024-01-24:
Sound good! Still have the sensor laying around but replaced it with a supported on at the time. So still would love support for it

kostecky, 2024-06-03:
@spattinson Could you please post your custom include? I'm trying to get this winsen zh03b working, but I can't get it to compile

spattinson, 2024-11-02:
I finally got around to doing this. There is an easier way than writing an external components, it can all be done in yaml: This is what I am using:

`esphome:
  name: zh03b
  platform: ESP32
  board: esp32dev
  on_boot:
    priority: 600
    then:
# command to enter Q&A mode
      - uart.write:  [0xFF, 0x01, 0x78, 0x41, 0x00, 0x00, 0x00, 0x00, 0x46]

logger:
  level: VERBOSE
api:
ota:
  platform: esphome

wifi:
  ssid: asus_c0
  password: *******

  ap:
    ssid: zh03b

captive_portal:

web_server:
  port: 80

uart:
  - id: uart_zh03b
    rx_pin: 13
    tx_pin: 15
    baud_rate: 9600
    debug:
      direction: RX
      dummy_receiver: true
      sequence:
        - lambda: |-
            UARTDebug::log_hex(direction, bytes,':');  //Log hex data
            if (bytes.size() == 9 && bytes[0] == 0xFF && bytes[1] == 0x86) {
              //only update if expected response bytes returned
              id(zh03b_pm1).publish_state( bytes[2] * 256 + bytes[3] );
              id(zh03b_pm2_5).publish_state( bytes[4] * 256 + bytes[5] );
              id(zh03b_pm10).publish_state( bytes[6] * 256 + bytes[7] );
            }

sensor:
  - platform: template
    name: PM1
    id: zh03b_pm1
    unit_of_measurement: ug/m3
    accuracy_decimals: 0
    lambda: "return {};"

  - name: PM2.5
    platform: template
    id: zh03b_pm2_5
    unit_of_measurement: ug/m3
    accuracy_decimals: 0
    lambda: "return {};"

  - name: PM10
    platform: template
    id: zh03b_pm10
    unit_of_measurement: ug/m3
    accuracy_decimals: 0
    lambda: "return {};"

interval:
  - interval: 1min
    then:
# exit dormant mode, start fan
      - uart.write:  [0xFF, 0x01, 0xA7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x58]
      - delay: 10s
# send question
      - uart.write:  [0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79]
      - delay: 2s
# enter dormant mode, stop fan
      - uart.write:  [0xFF, 0x01, 0xA7, 0x01, 0x00, 0x00, 0x00, 0x00, 0x57]`

djansen1987, 2024-11-13:
Great, thank you so much! i am getting values vor pm1,2.5 and 10.

Thank you for finding it and also sharing it!

closed 2024-11-13 by djansen1987