esphome/feature-requests #1558: AHT20 and AHT21 Temperature and Humidity Sensors

rothnic, 2021-12-21:
**Describe the problem you have/What new integration you would like**

I have been looking for quality cheap temperature/humidity sensors and found references to the [aht21](https://www.seeedstudio.com/blog/2020/10/29/aht21-the-new-generation-of-industrial-grade-temperature-and-humidity-sensor/). There is a project where someone has compared many sensors across environments and found that the aht21 (really the aht family in general) performs very well compared to more expensive sensors and appears much better than the dht counterparts, and are cheaper than the bme280. 

So, if you don't need barometric pressure, they seem like a better option. I was able to purchase 10 ah21 sensors through aliexpress for ~$14 shipped.

I am now looking at using esphome, since I use it for some other sensors, but it doesn't seem to support the aht21 sensor. It appears that support was added for [aht10](https://esphome.io/components/sensor/aht10.html?highlight=aht), but hasn't been updated for aht20 or aht21. The [library](https://github.com/Thinary/AHT10) referenced for the esphome implementation is 3 years old, while there is a [newer one](https://github.com/Thinary/AHT_Sensor) that supports all the aht sensors. I'm not sure if the implementation would need to be updated at this point, but the platform should be updated so that each of the aht models is explicitly supported.

**Please describe your use case for this integration and alternatives you've tried:**

Temperature and humidity data collection

**Additional context**

FalconUK, 2022-01-01:
I would also like support for the AHT20. I have used a number of  these with the NodeMCU ESP8266 and my own software, but it would be great if ESPhome supported them.

rothnic, 2022-01-01:
> I would also like support for the [AHT20](https://learn.adafruit.com/adafruit-aht20). I have used a number of these with the NodeMCU ESP8266 and my own software, but it would be great if ESPhome supported them.

I did end up getting my AHT21 sensor in and tested it using the AHT10 platform name in the esphome configuration. It seems to function just fine and appears to behave much better than my bme280.

So, an initial improvement would be to just update the documentation. If the implementation is the same for all of them, then it might make sense to have the platform name just be `aht`.

I'm still not sure what the difference is in the implementation linked in the different repositories though.

FalconUK, 2022-01-02:
Based on @rothnic's comments, I decided to just give it a try with a NodeMCU and an AHT20. It works fine.

I had a quick look at specifications etc and the main difference between the AHT10 and AHT20 appears to be the supply voltage range, so not anything that affects I2C communications. 

Also, Adafruit's arduino library is called 'AHTx0', which says something...

ChirpyTurnip, 2022-04-06:
The new library looks like it is providing generic support all AHT sensors....should definitely do it then.....supporting all models is better than one....just sayin'.....  :-)

sumirati, 2022-05-15:
I have created PR https://github.com/esphome/esphome-docs/pull/2081 to update the documentation.

nagyrobi, 2022-07-01:
Implemented: https://esphome.io/components/sensor/aht10.html?highlight=aht20

closed 2022-07-01 by nagyrobi