esphome/feature-requests #1559: Add `restore_state` to fans

nagyrobi, 2021-12-22:
**Describe the problem you have/What new integration you would like**

Please add `restore_state` to fans, specifically the speed fan and the binary fan.
Need to restore the on/off state, the speed, oscillating state.

**Please describe your use case for this integration and alternatives you've tried:**

Having a ventilation system commanded via slow_pwm relays or modbus lambdas in output components, impossible to read back the state of the unit after power cycle. Need to store the last state in flash.

Like it's possible for `select`.

**Additional context**

Thank you.

ssieb, 2022-01-10:
It already does this.  There's just no option to turn it off.

nagyrobi, 2022-01-10:
Yep, in the meanwhile it turned out that the vice-versa is what I need, to override this to OFF.
Sorry.

ssieb, 2022-01-10:
I suggest filing an issue for that instead of a feature request.

closed 2022-01-10 by ssieb

nagyrobi, 2022-01-10:
OK!