esphome/feature-requests #1560: SD Card as image storage for display

TheEris, 2021-12-23:
**Describe the problem you have/What new integration you would like**

The title says it all. I would love to have the option to put images (BMP, or converted image data, or ....) on an SD card and esphome would then read it and display it on display.

**Please describe your use case for this integration and alternatives you've tried:**

Huge UI. A lot of pages with fullscreen background and buttons/switches with multiple state images. Animated icons, etc...

**Additional context**

Currently, you can put only about 3~4 fullscreen (RGB24, ESP32) images in flash. If you could put the image data on SD card, you could have a huge UI with only a few lines of code.

beforeyouknowit, 2023-03-28:
+1 for this feature; ESPHome has support for a variety of TFT LCDs already, and not everything can be designed and drawn to screen efficiently with lambdas.

Papadoma, 2023-05-25:
+1 from me too. I want to try a holocube esphome version for weather forecast display and having an SD card makes it far more efficient

alfwro13, 2023-10-20:
alternative would be fetching them from a URL - like this proposal:
https://github.com/esphome/esphome/pull/4710

jamesmyatt, 2023-10-20:
Would be great to get to the point where you can implement the esp32-tv entirely in esphome: https://github.com/atomic14/esp32-tv