esphome/feature-requests #1563: allow update_interval for specific entities in modbus_controller

Bascht74, 2021-12-26:
**Describe the problem you have/What new integration you would like**
I am using modbus_controller for some time now and as I understand it, it polls the device to get all the states for the entities. 
for binary entities & switches - like on/off stuff - a fast update rate would be nice as for non-binary sensors - like temperature, humidity, power, energy, a slower rate is OK and saves resources on the HA side (because they change often)

**Please describe your use case for this integration and alternatives you've tried:**

I would like to set update_interval for specific entities or at least the entity component class.
At the end I get updates of switches, binary entities fast and normal sensors slow. That would be perfect for my modbus device that adds > 90 entities to HA.

**Additional context**

I don't know if this is easily possible, but I hope it is OK to ask.

Could this be a workaround or a solution?
1. Use a short update interval e.g. 10 seconds. 
2. Use a filter for all entities that are changed often and use throttle_average

martgras, 2021-12-27:
**TLDR**
use `skip_updates` 

**Details:**
Without changing the architecture of the modbus_controller component radically this it's not possible to use homor the update_interval for every modbus sensor.  
The component optimizes (reduces) modbus traffic by combining requests to consecutive addressess into ranges. Each range contains 1 or more sensor items 
If you define registers 0,1 and 2 the component will send the command "read 3 registers starting at address 0"
update_interval defined under `modbus_controller` defines the frequency of the range updates.
Therefore update_interval of the individual sensors is ignored

For my epever solar controller I get 16 ranges
```
[10:21:38][V][modbus_controller:126]: Range : 2 Size: 2 (1) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 5 Size: 2 (1) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 2000 Size: 1 (2) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 200C Size: 1 (2) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 9001 Size: 1 (3) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 9013 Size: 3 (3) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 3000 Size: 9 (4) skip: 60
[10:21:38][V][modbus_controller:126]: Range : 300E Size: 1 (4) skip: 60
[10:21:38][V][modbus_controller:126]: Range : 3100 Size: 8 (4) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 310C Size: 7 (4) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 311A Size: 2 (4) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 311D Size: 1 (4) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 3200 Size: 2 (4) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 3201 Size: 1 (4) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 3300 Size: 16 (4) skip: 0
[10:21:38][V][modbus_controller:126]: Range : 331A Size: 3 (4) skip: 0
```

Notice the `skip` field at the end of range output.  
While `update_interval` for the sensors is ingored there is 'skip_updates`
It can be used for values that don't change frequently. 
If set one of sensors in a range to a non-zero 'skip_updates` value it will be applied to the whole range. 
If more than one sensor in a range has 'skip'updates'  the lowest non-zero value in a range is used.
In the example below a range from registers 0x3000 - 0x3008 is created. Because of `skip_updates: 60`  all 9 registers are only read every 60th update interval 

yaml
```
sensor:  
  - platform: modbus_controller
    modbus_controller_id: epever
    id: array_rated_voltage
    name: "array_rated_voltage"
    address: 0x3000
    skip_updates: 60
    unit_of_measurement: "V"
    register_type: read
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      - multiply: 0.01

  - platform: modbus_controller
    modbus_controller_id: epever
    id: array_rated_current
    name: "array_rated_current"
    address: 0x3001
    unit_of_measurement: "A"
    register_type: read
    value_type: U_WORD
    accuracy_decimals: 2
    filters:
      - multiply: 0.01
... 
 - platform: modbus_controller
    modbus_controller_id: epever
    id: charging_mode
    name: "charging_mode"
    address: 0x3008
    unit_of_measurement: ""
    register_type: read
    value_type: U_WORD
    accuracy_decimals: 0
```

I can possibly implement `update_interval` similar to `skip_updates` applying to the complete range but I can't implement different `update_interval` values within a range of sensors. 
However even that requires several changes - not sure it's worth the effort given that `skip_updates` serves the same purpose. 
 Basically update_interval for an  item is calculated as ‘update_interval` defined for modbus_controller multiplied by skip_updates of a range.

closed 2021-12-27 by Bascht74

Bascht74, 2021-12-27:
Ok, I will try that (`skip_updates`) and `throttle_average`. 
Thx for the quick hint...