esphome/feature-requests #1564: New zigbee dev board, can be supported by esphome?

stevoh6, 2021-12-27:
**Describe the problem you have/What new integration you would like**

Open new world off posibilities for HA with new zigbee/ble5.0 dev board.
It is possible add support for this or similar board?
https://www.cnx-software.com/2021/12/27/xt-zb1-zigbee-ble-devkit-bl702-risc-v-module/

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

martgras, 2021-12-28:
I don't see support in the next future since this board uses a completely different toolchain/sdk than the esp modules.
The announced esp32-H2 will have similar features and has a much better chance for support.
However without wifi it is not that useful unless someone implements a ble gateway or some other alternative connectivity.

stevoh6, 2021-12-28:
Thanks, now after couple of other reactions/messages I understand why xt-zb is dead end. :)

Anyway this esp32-h2 sounds really promissing. Wel in theory you should have three options how to connect to esp (ble, usb, zigbee). For example zigbee2mqtt support ota ipdates, so ota updates should be fine. 

I can imagine how awesome it will be if a house/garden sensors will profit from zigbee mesh network, the possibilities will be endless. Yes it require tone of hard work.

Hedda, 2022-06-13:
See feature request for Zigbee on ESP32-H2 here -> https://github.com/esphome/feature-requests/issues/1397

Also see these indirectly related requests here -> https://github.com/esphome/feature-requests/issues/688 and https://github.com/esphome/feature-requests/issues/681

@stevoh6 The original feature request for XT-ZB1 support should be closed as not directly related to ESP8266 or ESP32