esphome/feature-requests #1567: Add the QMP6988 from the M5 Env III sensor

olaxe, 2021-12-28:
**Describe the problem you have/What new integration you would like**

Retrieve pressure from the QMP6988 embedded in the M5 Stack ENV III

**Please describe your use case for this integration and alternatives you've tried:**

I have tried to use the BMP280 but, of course, the chip id is not correct and I think other internal things are different

**Additional context**

The official documentation: https://docs.m5stack.com/en/unit/envIII
The QMP6988 datasheet: https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/unit/enviii/QMP6988%20Datasheet.pdf
The library for the chip: https://github.com/m5stack/UNIT_ENV/tree/master/src

comnam90, 2022-01-17:
+1 This would be a great addition

andrewpc, 2022-02-02:
+1 would be great. Thank you

alexruffell, 2022-02-18:
+1 The M5Stack Env III Sensor has a great packaging (except for the hard to find connectors) solution that makes it easier to use in projects. 

[image]

Support for the SHT30 is already present. All that is missing, for full support, is the QMP6988 pressure sensor support.

andrewpc, 2022-02-21:
No problems. Ive just updated the code based on the feedback so hopefully its ready to merge soon.

VentingPond, 2022-03-27:
+1.  Would be awesome to include this.  Please... :)

andrewpc, 2022-03-31:
This has now been merged and should be in the next major release. You can start using it now by referencing it on the dev branch as an external component:

```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome/
      ref: dev
    components: [ qmp6988 ]
```
Let me know if you experience any issues.

closed 2022-07-01 by nagyrobi