esphome/feature-requests #1568: Expose the `unavailable` state as configurable to all entities

nagyrobi, 2021-12-28:
**Describe the problem you have/What new integration you would like**
There should be a way to deliberately set the components state to `unavailable` programmatically from ESPHome.
Say, I have an external device I control with an ESP32. I have feedback from the external device via a GPIO pin set to high while it is normal operational state.
I'd like to set all the entities belonging to it to `unavailable` when this device dies for any reason (lets the GPIO pin drop).

**Please describe your use case for this integration and alternatives you've tried:**
With a binary sensor I'd collect the real state of the physical device, and when it's really, physically unavailable, set the desired components in esphome (and HA) to `unavailable`. A bunch of swithces, fans, selects, sensors etc. all belonging to the same main device should follow the availability state given by this binary sensor.

My suggestion is to add a templatable core binary property to _all_ components, which could be configured in yaml to follow the state of a binary sensor.

**Additional context**
This could be an extension of the way entities go unavailable in HA when the node disconnects.
This could also override that, as combined with https://github.com/esphome/feature-requests/issues/1492 one could even override the unavailable-on-disconnection behavior if he/she wants to.

nagyrobi, 2021-12-29:
Related [discord](https://discord.com/channels/429907082951524364/429907082955718657/925532426963398686).

nagyrobi, 2022-02-09:
Here's a new use case for this:

We have a ventilation system which can be operated in several ways (via modbus).
- one is manual mode, where you can select the fan speeds manually. This can be integrated either as a fan, or a select component
- one is automatic mode, where speeds are automatically selected by the unit based on he humidity of the absorbed air from inside (done by the ventilation system itsef with its own sensors). This can be integrated practically as a switch - just enable or disable this operation mode.

According to the tech spec of the machine, one should not adjust the speeds manually while the unit is placed in automatic mode.

The most elegant solution from the user's perspective would be to set the fan or speed select entitiy to `unavailable` state when the automatic mode is `ON`.

DJTerentjev, 2022-03-15:
I have same request. In my case I have custom components for rs485 device and I would like to set sensors to `unavailable` in case communication lost. I was really surprised this option doesn't exist.

github-k8n, 2022-05-06:
Same here, custom component for ventilation system where some sensors are optional (humidity sensors). Currently by sending NaN in the publish method in that case, the values are set to "unknown" which is good, but better would be if those sensors could be made "unavailable" to reflect the fact that they are in fact not present.

nagyrobi, 2022-06-20:
@OttoWinter maybe you have an opinion on this?

OttoWinter, 2022-06-20:
Please don't tag me into random discussions, I get enough notification spam

For sensors, I don't see why `NaN` aka `unknown` is an issue? Sure it looks a bit different, but :shrug:

nagyrobi, 2022-06-20:
They mean different things.
You can't set `unknown` state for a commanding component like a select or a number. 

We need `unavailable` too, to properly block users from using these when needed. Explained some use cases above.

lucius-the, 2023-01-25:
Another use case:
For my home alarm (most sensors are handled by esphome) I don't want to rely on WiFi, so I'm trying to use CAN. This seems to be the solution for me and initial tests look promising. But.

So I will have one esphome "master" node that will be connected via CAN to several other nodes. Now in case some node goes offline my master node will be able to know this (by timeout from last received message / not responding to status poll or whatever) but I would be happy to tell HA that some entities are now "Unavailable", to be in line with how everything else works in HA. But I don't see a way to do it currently.

DJTerentjev, 2023-05-29:
My user case: I have water sensor with connection control. Now I use two binary sensors. I was not able to combine in one (leak-dry) and mark  this sensor as `unavailable` as soon connection is lost. Even NAN is not possible to set for binary sensor. Please advise if any other way to do it.

kpfleming, 2023-08-23:
Home Assistant recently merged a [PR](https://github.com/home-assistant/core/pull/98110) which disallows NaN and Inf for numerical sensors. This likely means that starting with HA 2023.9 all of the components which use `publish_state(NAN)` to indicate 'unavailable' are going to be broken in HA.

kquinsland, 2023-08-23:
> Home Assistant recently merged a [PR](https://github.com/home-assistant/core/pull/98110) which disallows NaN and Inf for numerical sensors. This likely means that starting with HA 2023.9 all of the components which use `publish_state(NAN)` to indicate 'unavailable' are going to be broken in HA.

But critically, `unavailable` is [still supported](https://github.com/home-assistant/core/pull/98110#issuecomment-1690104630)

nagyrobi, 2023-08-23:
Oops we need something asap here

nagyrobi, 2023-08-24:
Don't know what other entity states HA supports apart from the usual device specific (binary, numeric values etc.) and `unavailable` or `unknown` but maybe it's a good time now to consider improving usability by implementing all existing.

functionpointer, 2023-12-08:
Turns out it doesn't even work with `sensor` anymore. Homeassistant throws an error and continues showing the last valid number.

https://github.com/home-assistant/core/pull/98110 is the reason, as @kpfleming has pointed out.

It still works if no unit of measurement is specified, Homeassistant then thinks the sensor is non-numeric. In that case nan is simply shown as a string.

nagyrobi, 2023-12-10:
> Please don't tag me into random discussions, I get enough notification spam
> 
> For sensors, I don't see why `NaN` aka `unknown` is an issue? Sure it looks a bit different, but 🤷

From now HA rejects NaN values.

clydebarrow, 2023-12-11:
> From now HA rejects NaN values.

Still working for me and the code seems to handle NaN specifically. Here's the relevant bit of HA code:

```
        # From here on a numerical value is expected
        numerical_value: int | float | Decimal
        if not isinstance(value, (int, float, Decimal)):
            try:
                if isinstance(value, str) and "." not in value and "e" not in value:
                    try:
                        numerical_value = int(value)
                    except ValueError:
                        # Handle nan, inf
                        numerical_value = float(value)       <--- "NaN" gets converted to the float value nan here.
                else:
                    numerical_value = float(value)  # type:ignore[arg-type]

```

Still kludgey, doesn't work for binary or text, and should be superseded by a proper mechanism to mark as unknown. But still working.

[image]

twasilczyk, 2024-05-06:
My suggestion is to have a separate API to set component availability - otherwise, components like Switch couldn't use it (it only accepts `publish_state(bool)`).

I think the best would be to add:
```
void set_available(bool available);
```
to `esphome::EntityBase`.

I'm happy to try implementing this, but would prefer to have some initial OK from esphome/homeassistant owners if this is the right approach. I guess I would need to:
- Add new message `EntityStatusResponse` (with field `available`) to `esphome/components/api/api.proto`. It would be sent to `subscribe_states` subscribers
- Add `set_available` to `EntityBase` and hook it up to `EntityStatusResponse` (probably through new `add_entity_status_callback`)
- Implement support for the new message to [aioesphomeapi](https://github.com/esphome/aioesphomeapi)
- Hook it up in esphome HA integratinn

Am I missing anything? Any pitfalls?

nagyrobi, 2024-06-05:
Please go ahead, we need this.

functionpointer, 2024-06-07:
Is this related? https://github.com/home-assistant/core/pull/117813

nagyrobi, 2024-06-07:
Nope. It's MQTT-only.

Stipena, 2024-06-23:
I also need function for switch and number components to be greyed out if like example binary sensor is off.

AJolly, 2025-03-21:
this would be useful

yawor, 2025-07-23:
This would be very useful. Marking entities as unavailable is a standard way of informing user in HA that the device/entity has become, well, unavailable. It may be temporarily, for example because of the power outage or some temporary communication issues.

Given the recently added devices (subdevices) support, it would be even better to have this exposed on the device level, so all entities connected to device marked as unavailable would also become unavailable.

nagyrobi, 2025-07-23:
Binary sensor is the first one to support this.