esphome/feature-requests #1570: Inkplate 6PLUS

JMillwood-92, 2021-12-30:
Hello,

I have been experimenting with the inkplate 6 plus for a few days now. 

I noticed that ESPHome supports inkplate 6 & 10, and was wondering if the 6 PLUS will be added to the compatible models. 

Best Wishes,
Josh.

MickazNL, 2022-01-02:
Would love inkplate 6 plus support as well. I've had contact with the developer and support should be on it's way although I haven't heard from them in a while.

jjhuff, 2022-01-06:
Was that @jesserockz ? I have a 6plus and have partially implemented the changes, so i'd be happy to collaborate.

MickazNL, 2022-01-06:
No I don't think so. I have a 6 plus as well which I bought for use with esphome and Home Assistant.

Tried to load the Inkplate 6 esphome image on it but it only got it started up once and it's beyond my skillset to change the image.

JMillwood-92, 2022-01-06:
I've managed to edit the original inkplate6.h file to take advantage of the slightly larger screen on the 6 plus. But it's the touch functionality that I really want to utilise. 

At the moment my without ESPHome plan is to write an Electron App that creates a screen using html and information retrieved using MQTT, this gets exported to a jpeg that arduino code on the inkplate will pull and display. Then all touch presses are sent via MQTT and interpreted by homeassistant to identify if any action needs to be triggered, keep the inkplate alive long enough to receive either a no action or an update screen message then deep sleep until the next touch event.

All this seems a tad ott and involved if ESPHome will eventually support the 6 plus. 
Will also loose the nice integrated feel that ESPHome has with the integrated devices and entities offered. 

Apologies for the waffle, but thought I'd give my planned approach, in case anyone else has some better work around a whilst official support is added.

jesserockz, 2022-01-06:
Ill just leave this here: https://github.com/esphome/esphome/pull/3013

I am going to work on the touch part next

jesserockz, 2022-01-06:
This is also for the backlight control with brightness: https://github.com/esphome/esphome/pull/3014

jjhuff, 2022-01-06:
sweet! I'll give it a try tonight.

JMillwood-92, 2022-01-06:
Nice 👌 will have a look when I'm off next.

MickazNL, 2022-01-11:
Sorry to ask, I couldn't wait so I wanted to try it out. But I don't know how to use esphome with the files you changed / updated. I changed the github branch and took the Inkplate6 components folder then uploaded it to my docker installation. The behaviour for the device is the same as it was last time for me. Do I need a different installation yaml file?

Sorry to bother, else I will have to wait a little longer.

jjhuff, 2022-01-11:
@MickazNL I checked out the branch and then built the docker container from that:
`./docker/build.py --arch amd64 --build-type docker --tag inkplate-6-plus build --load`

Then, just use `esphome/esphome-amd64:inkplate-6-plus` in your `docker run` commands.

jesserockz, 2022-01-12:
@MickazNL @jjhuff 
Adding this to your yaml should pull the PR code to use with no other changes needed

```yaml

external_components:
  - source: github://pr#3013
    components: ["inkplate6"]
```

mkocus, 2022-01-19:
Hi @jesserockz,

I'm trying to use your PR on Inkplate 6 plus. I have the default config taken from here:
https://esphome.io/components/display/inkplate6.html
plus I've added your `external_components` declaration at the bottom.

Do I have to change anything? Platform / display / board etc?

jesserockz, 2022-01-19:
@mkocus Yes, add `model: INKPLATE_6_PLUS`

mkocus, 2022-01-19:
Thanks! Now the screen looks better :)

Do you have plans/timeline for adding support for touchscreen?

jesserockz, 2022-01-20:
@mkocus that pr is open too

mkocus, 2022-01-20:
@jesserockz I'm testing your PR for display and it seems to generally work (apart from the partial update issue, but it is not connected).

Have you managed to get the proper battery voltage readings? When I'm using the config from Inkplate 6 example it shows:
```
[D][sensor:125]: 'Inkplate Battery Voltage': Sending state 0.28400  with 1 decimals of accuracy
```
which is of course not correct.

But running the Inkplate_Battery_Voltage_And_Temperature example for Arduino IDE shows proper voltage.

jesserockz, 2022-01-20:
@mkocus I have yet to run my 6plus with a battery so cannot comment on that part, but I will take a look when I get a chance.

mkocus, 2022-01-26:
@jesserockz I've managed to get the proper battery voltage readings by directly converting the Arduino code into the ESPHome. Only for test purpose I've change the visibility of set_pin_mode method of gpio switch.

But I don't know how to do it nicely - is there a proper way to change the pin mode (input/output) of the mcp23017 port extender?

That's my config for that. battery_read_mosfet is switch/gpio declared as mcp23017 pin.
```
  - platform: template
    name: "Inkplate Battery Voltage"
    accuracy_decimals: 3
    unit_of_measurement: "V"
    lambda: |-
      GPIOPin* pin = id(battery_read_mosfet).get_pin();
      pin->pin_mode(gpio::FLAG_INPUT);
      int state = pin->digital_read();
      pin->pin_mode(gpio::FLAG_OUTPUT);
      if (state) {
        pin->digital_write(false);
      } else {
        pin->digital_write(true);
      }
      delay(1);
      id(battery_voltage)->set_output_raw(true);
      int adc = id(battery_voltage).sample();
      if (state){
        pin->pin_mode(gpio::FLAG_INPUT);
      } else {
        pin->digital_write(false);
      }
      return (double(adc) / 4095 * 1.1 * 3.548133892 * 2);
```
And here is your original lambda for that (but it always read the same, wrong adc value):
```
id(battery_read_mosfet).turn_on();
delay(1);
float adc = id(battery_voltage).sample();
id(battery_read_mosfet).turn_off();
return adc;
```

jesserockz, 2022-01-28:
The returned `adc` value is not the actual battery voltage. It is the adc pin voltage which is the battery voltage divided by the resistors (*2 to get back to original voltage in your code) but I have found that its not quite (*2) on my Inkplate10. I need to plug a battery in and try the 6Plus to see what values I get, but you should only need to use a simple `calibrate_linear` filter to convert the adc value into the correct measured battery voltage.

mkocus, 2022-01-30:
Yes, noticed that. But in Inkplate 6plus case, I always get the same reading regardless of real battery voltage (0.28V), so filtering the data would not case (of course at least in my case). It seems that changing the pin mode + writing to pin before sampling helped.

closed 2022-02-08 by jesserockz

shokwaav, 2022-02-26:
How do I integrate the backlight into the .yaml?

jesserockz, 2022-02-27:
@shokwaav 
```yaml
i2c:

power_supply:
  - id: backlight_power
    keep_on_time: 0.2s
    enable_time: 0s
    pin:
      mcp23xxx: mcp23017_hub
      number: 11

output:
  - platform: mcp47a1
    id: backlight_brightness_output
    power_supply: backlight_power

light:
  - platform: monochromatic
    output: backlight_brightness_output
    id: backlight
    default_transition_length: 0.2s
    name: '${friendly_name} Backlight'
```

shokwaav, 2022-02-27:
@jesserockz that worked perfectly thank you!

MickazNL, 2022-03-05:
Everything is working fine thanks for all the effort!

I only can't seem to get the partial update working as I expected. For example I'm dispaying a clock and only the minutes update. Then the rest of the screen is getting dimmed. Is this expected behaviour of the device or am I doing something wrong?

Below my display properties, please don't mind my lambda i'm a noob. I'm trying to create a dashboard for Home Assistant with for now; time, temperature and co2 values of the livingroom.

```
display:
- platform: inkplate6
  model: INKPLATE_6_PLUS
  id: inkplate_display
  greyscale: false
  partial_updating: true
  update_interval: 60s
  full_update_every: 10

  ckv_pin: 32
  sph_pin: 33
  gmod_pin:
    mcp23xxx: mcp23017_hub
    number: 1
  gpio0_enable_pin:
    mcp23xxx: mcp23017_hub
    number: 8
  oe_pin:
    mcp23xxx: mcp23017_hub
    number: 0
  spv_pin:
    mcp23xxx: mcp23017_hub
    number: 2
  powerup_pin:
    mcp23xxx: mcp23017_hub
    number: 4
  wakeup_pin:
    mcp23xxx: mcp23017_hub
    number: 3
  vcom_pin:
    mcp23xxx: mcp23017_hub
    number: 5

  lambda: |-
    it.fill(COLOR_ON);

    it.print(100, 100, id(helvetica_48), COLOR_OFF, TextAlign::TOP_LEFT, "ESPHome");

    it.strftime(400, 300, id(helvetica_48), COLOR_OFF, TextAlign::CENTER, "%d-%m-%Y", id(esptime).now());
    it.strftime(400, 400, id(helvetica_96), COLOR_OFF, TextAlign::CENTER, "%H:%M", id(esptime).now());

    if (id(system_status).state) {
      it.print(700, 100, id(helvetica_48), COLOR_OFF, TextAlign::TOP_RIGHT, "Online");
    } else {
      it.print(700, 100, id(helvetica_48), COLOR_OFF, TextAlign::TOP_RIGHT, "Offline");
    }
        it.image(100, 500, id(thermo));
    if (id(woonkamer_lucht_temperature).has_state()) {

    it.printf(150, 500, id(helvetica_96), COLOR_OFF, TextAlign::CENTER, "%.1f", id(woonkamer_lucht_temperature).state);
    }
    if (id(woonkamer_co2).has_state()) {
    it.printf(700, 500, id(helvetica_96), COLOR_OFF, TextAlign::CENTER, "%.0f", id(woonkamer_co2).state);
    }
```

mkocus, 2022-03-05:
Unfortunately this is an expected behavior on Inkplate 6plus.

As a matter of fact in my opinion this is actually the only most important
issue I have with this device.

jesserockz, 2022-03-11:
Unfortunately this happens to almost every e-paper display, especially when the "drivers" are reverse engineered and not specifically built by the manufacturer like on the Inkplate series.

jjhuff, 2022-04-04:
> Yes, noticed that. But in Inkplate 6plus case, I always get the same reading regardless of real battery voltage (0.28V), so filtering the data would not case (of course at least in my case). It seems that changing the pin mode + writing to pin before sampling helped.

@mkocus I actually got it working by swapping the on/off commands. Not sure if un-inverting would fix it too -- but I wasn't sure if that's affect the initial state.

ADeadPixel, 2022-06-28:
Sorry for reviving a dead/stale issue thread but idk where else to ask this. How do I implement the touchscreen into the config? I followed the example in the documentation and it gives an arguments error. So I stripped out the args and just left a simple log message but that does not show up anywhere. 

Here is my config with the touchscreen section at the bottom. Any help/guidance would be appreciated!
`esphome:
  name: inkplate

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Inkplate Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

captive_portal:

switch:
  - platform: restart
    name: "Inkplate Reboot"
    id: reboot
    
  - platform: gpio
    id: battery_read_mosfet
    pin:
      mcp23xxx: mcp23017_hub
      number: 9
      inverted: true

  - platform: template
    name: "Inkplate Greyscale mode"
    lambda: return id(inkplate_display).get_greyscale();
    turn_on_action:
      - lambda: id(inkplate_display).set_greyscale(true);
    turn_off_action:
      - lambda: id(inkplate_display).set_greyscale(false);

  - platform: template
    name: "Inkplate Partial Updating"
    lambda: return id(inkplate_display).get_partial_updating();
    turn_on_action:
      - lambda: id(inkplate_display).set_partial_updating(true);
    turn_off_action:
      - lambda: id(inkplate_display).set_partial_updating(false);
      
  - platform: template
    name: "Inkplate Refresh Screen"
    icon: "mdi:refresh"
    internal: false
    turn_on_action:
      - lambda: id(inkplate_display).update();

sensor:
  - platform: adc
    id: battery_voltage
    update_interval: never
    attenuation: 11db
    pin: 35
  - platform: template
    name: "Inkplate Battery Voltage"
    lambda: |-
      id(battery_read_mosfet).turn_on();
      delay(1);
      float adc = id(battery_voltage).sample();
      id(battery_read_mosfet).turn_off();
      return adc;
    filters:
      - multiply: 2
      
text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
      id: wifi_ip_address
    ssid:
      name: ESP Connected SSID
      id: wifi_ssid
    bssid:
      name: ESP Connected BSSID
      id: wifi_bssid
    mac_address:
      name: ESP Mac Wifi Address
      id: wifi_mac
      
i2c:

power_supply:
  - id: backlight_power
    keep_on_time: 0.2s
    enable_time: 0s
    pin:
      mcp23xxx: mcp23017_hub
      number: 11

output:
  - platform: mcp47a1
    id: backlight_brightness_output
    power_supply: backlight_power

light:
  - platform: monochromatic
    output: backlight_brightness_output
    id: backlight
    default_transition_length: 0.2s
    name: 'Inkplate Backlight'

mcp23017:
  - id: mcp23017_hub
    #address: 0x20

binary_sensor:
  - platform: status
    name: "Inkplate Status"
    id: system_status

time:
  - platform: homeassistant
    id: esptime

font:
  - file: "Roboto-Regular.ttf"
    id: RobotoRegular_96
    size: 96
  - file: "Roboto-Regular.ttf"
    id: RobotoRegular_48
    size: 48
  - file: "Roboto-Regular.ttf"
    id: RobotoRegular_24
    size: 24

display:
- platform: inkplate6
  id: inkplate_display
  greyscale: false
  partial_updating: false
  update_interval: 60s
  model: inkplate_6_plus

  ckv_pin: 32
  sph_pin: 33
  gmod_pin:
    mcp23xxx: mcp23017_hub
    number: 1
  gpio0_enable_pin:
    mcp23xxx: mcp23017_hub
    number: 8
  oe_pin:
    mcp23xxx: mcp23017_hub
    number: 0
  spv_pin:
    mcp23xxx: mcp23017_hub
    number: 2
  powerup_pin:
    mcp23xxx: mcp23017_hub
    number: 4
  wakeup_pin:
    mcp23xxx: mcp23017_hub
    number: 3
  vcom_pin:
    mcp23xxx: mcp23017_hub
    number: 5

  lambda: |-
    it.fill(COLOR_ON);

    it.print(50, 100, id(RobotoRegular_48), COLOR_OFF, TextAlign::TOP_LEFT, "BerryHaus Dashboard");

    it.strftime(512, 500, id(RobotoRegular_48), COLOR_OFF, TextAlign::CENTER, "%m-%d-%Y", id(esptime).now());
    it.strftime(512, 600, id(RobotoRegular_96), COLOR_OFF, TextAlign::CENTER, "%H:%M", id(esptime).now());

    if (id(system_status).state) {
      it.print(950, 100, id(RobotoRegular_48), COLOR_OFF, TextAlign::TOP_RIGHT, "Online");
      it.print(50, 730, id(RobotoRegular_24), COLOR_OFF, TextAlign::TOP_LEFT, "Online");
    } else {
      it.print(950, 100, id(RobotoRegular_48), COLOR_OFF, TextAlign::TOP_RIGHT, "Offline");
      it.print(50, 730, id(RobotoRegular_24), COLOR_OFF, TextAlign::TOP_LEFT, "Offline");
    }
    
    it.printf(1020, 730, id(RobotoRegular_24), COLOR_OFF, TextAlign::TOP_RIGHT, "ip address %s", id(wifi_ip_address).state.c_str());
  
touchscreen:
  - platform: ektf2232
    interrupt_pin: GPIO36
    rts_pin:
      mcp23xxx: mcp23017_hub
      number: 10
    on_touch:
      - logger.log:
          format: "touch detected"
`

ADeadPixel, 2022-06-28:
Solved the issue! If anyone finds this down the line you need to enable the touchscreen power by switching on pin 12 on the mcp23017. See coding example below found on https://github.com/esphome/esphome/pull/3027

```
switch:
  - platform: gpio
    name: 'Inkplate Touchscreen Enabled'
    restore_mode: ALWAYS_ON
    pin:
      mcp23xxx: mcp23017_hub
      number: 12
      inverted: true

touchscreen:
  - platform: ektf2232
    interrupt_pin: GPIO36
    rts_pin:
      mcp23xxx: mcp23017_hub
      number: 10
    on_touch:
      - logger.log:
          format: "touch x=%d, y=%d"
          args: ['touch.x', 'touch.y']
```