esphome/feature-requests #1571: Support for AwoX Mesh Lights

ahpohl, 2021-12-30:
**Describe the problem you have/What new integration you would like**

Add support for AwoX / Eglo Mesh Lights 

**Please describe your use case for this integration and alternatives you've tried:**

I would like to control AwoX and Eglo bluetooth lights from ESPHome. The custom component should add MQTT and HA api endpoints, from which the lights can be controlled.

There are already a Home Assistant custom component and a python library, which could serve as a basis:
Home Assistant [AwoX component](https://github.com/fsaris/home-assistant-awox) by @fsaris 
Python library [python-awox-mesh-light](https://github.com/Leiaz/python-awox-mesh-light) by @leiaz

**Additional context**

The advantage of controlling the lights from ESPHome and not directly from the home automation system would be that the ESP devices are often located closer to the lights, whereas the Home Assistant server is hidden away somewhere out of range of the lights. A new ESPHome component would also make it possible to control the lights with _any_ automation system which has MQTT support (Home Assistant, OpenHAB, etc ...)  

Potential devices (from the HA custom component):
- Eglo (120) RGBW spots (ESpot_120 HW v4.3)
- LED strip 3m (HW v4.0 and v4.3)
- LED strip 5m (HW v4.3)
- ECeil_G60 (HW v4.0)
- EPanel_120 (HW v4.0)
- EPanel_600 (HW v4.0)
- Eglo Fueva-C RGB CCT
- Eglo Crossling RGBW Spot (ETriSpot 85 HW 4.31, firmware 2.2.6)
- Eglo Crossling RGBW Spot (ESpot 170 HW 4.30, firmware 2.2.6)
- Eglo Connect Plug Plus (on/off support over Bluetooth)

fsaris, 2022-01-08:
Ah, nice idea. 

Anyone a suggestion for some hardware I could purchase to give it a try to connect to the ble mesh using esphome as base.

ahpohl, 2022-01-09:
I would like to control this one: [Eglo Connect Fueva-C](https://www.eglo.com/de/aufbauleuchte-fueva-c-96679.html) LED panel for the ceiling. But I guess any bulb or lamp from the Eglo/AwoX Connect range with bluetooth will do.

After integrating the Fueva-C into the AwoX clould with the iPhone app I am able to control the lamp with Home Assistant. Thanks for creating the custom component for this. I played around a bit with the python library, but didn't get very far. I created an ESPHome firmware with the ble tracker and ble client and was able to connect and get the MAC address this way:

```
esphome:
  name: espplay
  platform: ESP32
  board: esp32cam

wifi:
  ssid: "MQTT"
  password: "your-secure-wifi-password"

logger:
  level: VERY_VERBOSE

ota:

web_server:
  port: 80

mqtt:
  broker: 192.168.178.55
  username: esphome
  password: "your-secure-mqtt-password"

esp32_ble_tracker:
  on_ble_advertise:
    - mac_address: A4:C1:38:B3:0D:79
      then:
        - lambda: |-
            ESP_LOGD("ble_adv", "New BLE device");
            ESP_LOGD("ble_adv", "  address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  name: %s", x.get_name().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
                ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
            ESP_LOGD("ble_adv", "  Advertised service data:");
            for (auto data : x.get_service_datas()) {
                ESP_LOGD("ble_adv", "    - %s: (length %i)", data.uuid.to_string().c_str(), data.data.size());
            }
            ESP_LOGD("ble_adv", "  Advertised manufacturer data:");
            for (auto data : x.get_manufacturer_datas()) {
                ESP_LOGD("ble_adv", "    - %s: (length %i)", data.uuid.to_string().c_str(), data.data.size());
            }

ble_client:
  - mac_address: A4:C1:38:B3:0D:79
    id: ble_awox_light
    on_connect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Connected to BLE device AwoX light");
``` 
And this is the response in the log file:
```[VV][esp32_ble_tracker:494]: Parse Result:
[VV][esp32_ble_tracker:511]:   Address: A4:C1:38:B3:0D:79 (PUBLIC)
[VV][esp32_ble_tracker:513]:   RSSI: -75
[VV][esp32_ble_tracker:514]:   Name: ''
[VV][esp32_ble_tracker:522]:   Ad Flag: 6
[VV][esp32_ble_tracker:528]:   Manufacturer data: 02.15.43.2F.53.49.6E.8D.48.06.95.7A.D3.F9.73.5A.C7.1B.01.60.10.40.BA (23)
[VV][esp32_ble_tracker:531]:     iBeacon data:
[VV][esp32_ble_tracker:532]:       UUID: 1BC75A73-F9D3-7A95-0648-8D6E49532F43
[VV][esp32_ble_tracker:533]:       Major: 352
[VV][esp32_ble_tracker:534]:       Minor: 4160
[VV][esp32_ble_tracker:535]:       TXPower: -70
[VV][esp32_ble_tracker:544]: Adv data: 02.01.06.1A.FF.4C.00.02.15.43.2F.53.49.6E.8D.48.06.95.7A.D3.F9.73.5A.C7.1B.01.60.10.40.BA (30)
[D][ble_adv:074]: New BLE device
[D][ble_adv:075]:   address: A4:C1:38:B3:0D:79
[D][ble_adv:076]:   name: 
[D][ble_adv:077]:   Advertised service UUIDs:
[D][ble_adv:081]:   Advertised service data:
[D][ble_adv:085]:   Advertised manufacturer data:
[D][ble_adv:087]:     - 0x004C: (length 23)
[D][ble_client:047]: Found device at MAC address [A4:C1:38:B3:0D:79]
[V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.12 s).
[V][component:200]: Components should block for at most 20-30ms.
[I][ble_client:085]: Attempting BLE connection to a4:c1:38:b3:0d:79
[V][ble_client:115]: [a4:c1:38:b3:0d:79] ESP_GATTC_OPEN_EVT
[D][ble_client_lambda:096]: Connected to BLE device AwoX light
[VV][scheduler:185]: Running interval '' with interval=10000 last_execution=251556 (now=261560)
[V][ble_client:134]: cfg_mtu status 0, mtu 23
[V][ble_client:159]: [a4:c1:38:b3:0d:79] ESP_GATTC_SEARCH_CMPL_EVT
[I][ble_client:161]: Service UUID: 0x1800
[I][ble_client:162]:   start_handle: 0x1  end_handle: 0x6
[I][ble_client:341]:  characteristic 0x2A00, handle 0x3, properties 0x2
[V][ble_client:376]:    descriptor 0x2901, handle 0x4
[I][ble_client:341]:  characteristic 0x2A01, handle 0x6, properties 0x2
[I][ble_client:161]: Service UUID: 0x180A
[I][ble_client:162]:   start_handle: 0x7  end_handle: 0xf
[I][ble_client:341]:  characteristic 0x2A26, handle 0x9, properties 0x2
[I][ble_client:341]:  characteristic 0x2A29, handle 0xb, properties 0x2
[I][ble_client:341]:  characteristic 0x2A24, handle 0xd, properties 0x2
[I][ble_client:341]:  characteristic 0x2A27, handle 0xf, properties 0x2
[I][ble_client:161]: Service UUID: 00010203-0405-0607-0809-0A0B0C0D1910
[I][ble_client:162]:   start_handle: 0x10  end_handle: 0x1c
[I][ble_client:341]:  characteristic 00010203-0405-0607-0809-0A0B0C0D1911, handle 0x12, properties 0x1a
[V][ble_client:376]:    descriptor 0x2901, handle 0x13
[I][ble_client:341]:  characteristic 00010203-0405-0607-0809-0A0B0C0D1912, handle 0x15, properties 0xe
[V][ble_client:376]:    descriptor 0x2901, handle 0x16
[I][ble_client:341]:  characteristic 00010203-0405-0607-0809-0A0B0C0D1913, handle 0x18, properties 0x6
[V][ble_client:376]:    descriptor 0x2901, handle 0x19
[I][ble_client:341]:  characteristic 00010203-0405-0607-0809-0A0B0C0D1914, handle 0x1b, properties 0xa
[V][ble_client:376]:    descriptor 0x2901, handle 0x1c
```
It shows the MAC address, the service UUIDs and the ble characteristics, but these are already known. I didn't get any further than this.

fsaris, 2023-02-07:
@ahpohl I made a first version see https://github.com/fsaris/EspHome-AwoX-BLE-mesh-hub