esphome/feature-requests #1574: Ability to specify a default value for substitutions

bdr99, 2022-01-02:
**Describe the problem you have/What new integration you would like**
It would be helpful if we could provide a default value for substitutions. For example:
```yaml
esphome:
  name: ${device_id}
  platform: ESP8266
  board: esp8285
  esp8266_restore_from_flash: ${restore_from_flash:true} # This would default to true if 'restore_from_flash' is not defined
```

**Please describe your use case for this integration and alternatives you've tried:**
I have many devices which all use the same common base config. Each device-specific config file uses `!include` to import the common base config, and then uses substitutions to set device-specific config values such as device name and static IP address. I would like be able to override specific elements of the core config via a substitution on a per-device basis, without having to include that substitution in all of the devices.

For example, I may want `esp8266_restore_from_flash` set to true for all of my devices except one. Currently this would require me to create a substitution and include it in every device-specific config file, just so I can override it for that one device. The ability to specify a default would allow me to include this substitution only in the one device where I want to override the value.

**Additional context**

mkaatman, 2022-01-04:
Have you looked at packages? https://esphome.io/guides/configuration-types.html#packages

ThreeFN, 2022-01-11:
I was just investigating this.

Documentations states:
_**Packages**
Another way to modularize and reuse your configuration is to use packages. This feature allows you to put common pieces of configuration in separate files and keep only unique pieces of your config in the main yaml file. All definitions from packages will be merged with your main config in non-destructive way so **you could always override some bits and pieces of package configuration.**_

Though doesn't reference how to actually go about doing said overriding.

I found the following (presumed hacky) workaround where you define the substitution both within the package and at the device config main level.  The main level config substitutions value will override the one within the package.  See below working example where **update_when** is overridden.

Maybe it would be helpful if this sort of information could be included in the documentation, either in packages or substitutions, whichever is preferred.

```
# elliptical.yaml device configuration
substitutions:
  devicename: elliptical
  friendly_name: Elliptical
  update_when: 30s

packages:
  device_basis: !include device_configs/sonoff_s31.yaml
```

```
# in device_configs/sonoff_s31.yaml
substitutions:
  update_when: 3s

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

packages:
  base: !include base.yaml

logger:
  baud_rate: 0 # (UART logging interferes with cse7766)

# Device Specific Config
uart:
  rx_pin: RX
  baud_rate: 4800

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "${friendly_name} Button"
    on_press:
      - switch.toggle: relay
  - platform: status
    name: "${friendly_name} Status"

sensor:
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s
  - platform: cse7766
    current:
      name: "${friendly_name} Current"
      accuracy_decimals: 1
    voltage:
      name: "${friendly_name} Voltage"
      accuracy_decimals: 1
    power:
      name: "${friendly_name} Power"
      accuracy_decimals: 1
    update_interval: $update_when
switch:
  - platform: gpio
    name: "${friendly_name} Relay"
    pin: GPIO12
    id: relay

status_led:
  pin: GPIO13
```

```
# in device_configs/base.yaml
# base level information for all esphome devices

wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_password

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${friendly_name} AP
    password: !secret esphome_fallback_hotspot_password

captive_portal:

logger:

# Enable Home Assistant API
api:
  password: !secret esphome_api_password
  reboot_timeout: 1h

ota:
  password: !secret esphome_update_password
```

bdr99, 2022-01-12:
I agree with @ThreeFN. The packages feature seems useful, but the documentation is unclear. Even if it is possible to override/merge multiple configs with packages, I think substitutions have their own uses which cannot be accomplished with packages, such as including multiple substitutions within a single string. Therefore, I still think it would be useful to be able to specify a default value for substitutions.

closed 2022-07-01 by nagyrobi