esphome/feature-requests #1576: RC522 Gain control configuration

egroen, 2022-01-03:
**Describe the problem you have/What new integration you would like**
Adding a configuration option for the RC522 to change the gain (`RF_CFG_REG`)

**Please describe your use case for this integration and alternatives you've tried:**
When the default gain isn't viable, it can be increased for better penetration

I've manually added this locally to increase range through thin wooden wall for a small receiver tag, lost code due to update
But in fact nothing more than a `pcd_write_register(RF_CFG_REG, [ConfigValue]);` on the proper location

**Additional context**
It's a small addition, where default value is 0x48
So possible config value 0-7, default 4. Register = ((reg&0x07) << 4 ) | 0x08
Register layout:
```
bit: Desc        Default
  7: Reserved    [b0]
6-4: RxGain      [b100]
3-0: Reserved    [b1000]
```

egroen, 2022-01-03:
Currently I've got it working through config, but the validator is bugging me 'invalid value for [0] check indentation'
It's my first time playing with that stuff

Config
```yaml
rc522_spi:
  gain: 6
  ...
```
rc522.cpp:
```cpp
+@95: pcd_write_register(RF_CFG_REG, ((gain_&0x07) << 4) | 0x08);
```

rc522.h:
```hpp
uint8_t gain_{4};
void set_gain(uint8_t gain) { this->gain_ = gain; }
```

\_\_init\_\_.py:
```python
+@27: cv.Optional(CONF_GAIN, default=4): cv.All(cv.uint8_t, cv.Range(min=0, max=7)),
+@45: cg.add(var.set_gain(config[CONF_GAIN]));
```

selamanse, 2024-04-14:
@nagyrobi can someone have a look at my PR #6063 please? I have tried to implement the requested changes.

selamanse, 2025-02-01:
@nagyrobi anyone? still waiting for final review esphome/esphome#7539