esphome/feature-requests #1577: Boiler controller with a feeder SKZP-02

sargus123, 2022-01-03:
**Describe the problem you have/What new integration you would like**

Reading data from the furnace controller using UART RS232 communication such as: temperatures, furnace status, settings, alarms controller from the SKZP-02 family.
https://costerźniki.pl/pl/p/Sterownik-kotla-SKZP-02E/11

**Please describe your use case for this integration and alternatives you've tried:**

 I was able to connect to the UART and read the data packet (debug).
  I can not edit them as data to be displayed in the Homeassistamt (as well as data from a solar inverter)
If I had a few lines of code (files) reading the data below, I would be happy to complete the integration of the rest of the items ....

**Additional context**

Below I present the configuration I have and the data I have in the preview.

captive_portal:
# Example configuration entry
uart:
  id: uart_bus
  tx_pin: 17
  rx_pin: 16
  baud_rate: 57600
#  debug:
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
text_sensor:
- platform: custom
  lambda: |-
    auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
    App.register_component(my_custom_sensor);
    return {my_custom_sensor};
  text_sensors:
    id: "uart_readline"

dline': Sending state '{"DevId":"TASI7 S8","DevPin":"TASI7 OKN","Token":"GQMMW","FrameType":"SkzpData","TimeStamp":"22-05-06","BoilerTempAct":"4542","AN01":"1410","CH1ReturnTempAct":"3716","CH1ReturnTem'
[21:55:40][D][uart_debug:158]: <<< "\"K006\":\"1\",\"K007\":\"0\"}\r\n"
[21:55:41][D][uart_debug:158]: <<< "{\"DevId\":\"TASI7 S8\",\"DevPin\":\"TASI7 OKN\",\"Token\":\"GQMMW\",\"FrameType\":\"SkzpData\",\"TimeStamp\":\"22-05-10\","
[21:55:41][D][uart_debug:158]: <<< "\"BoilerTempAct\":\"4542\",\"AN01\":\"1420\",\"CH1ReturnTempAct\":\"3718\",\"CH1ReturnTempCmd\":\"3000\",\"CH1ReturnProtAct\":\"On\","
[21:55:41][D][uart_debug:158]: <<< "\"BoilerTempCmd\":\"5000\",\"DevStatus\":\"\",\"Alarms\":\"00000000I0K00000000000\",\"DHWTempAct\":\"3016\",\"DHWTempCmd\":\"7800\","
[21:55:42][D][uart_debug:158]: <<< "\"DHWHist\":\"500\",\"DHWMode\":\"Stop\",\"DHWPriority\":\"Off\",\"DHWOverH\":\"1000\",\"UpTime\":\"0d4h12m52s\","
[21:55:42][D][uart_debug:158]: <<< "\"CH1MixTempAct\":\"30000\",\"CH1MixTempCmd\":\"0\",\"CH1MixTempBase\":\"7500\",\"CH1MixTempMin\":\"2500\",\"CH1MixTempMax\":\"8000\","
[21:55:42][D][uart_debug:158]: <<< "\"C046\":\"0\",\"CH1MixValueAct\":\"0\",\"CH1MixValueMin\":\"10\",\"CH1MixValueMax\":\"100\",\"CH1MixGain\":\"10\","
[21:55:42][D][uart_debug:158]: <<< "\"CH1MixPeriod\":\"20\",\"CH1MixActive\":\"Off\",\"CH1RoomTempAct\":\"2919\",\"CH1RoomTempCmd\":\"9999\",\"CH1RoomTempEco\":\"1900\","
[21:55:42][D][uart_debug:158]: <<< "\"CH1RoomTempCom\":\"2100\",\"CH1RoomHist\":\"50\",\"CH1Mode\":\"Active\",\"CH1RoomMode\":\"Off\",\"CH2Mode\":\"Stop\","
[21:55:42][D][uart_debug:158]: <<< "\"ExhaustTempAct\":\"4333\",\"ExhaustTempMax\":\"25000\",\"WeaTempAct\":\"30000\",\"WeaTempStopCH1\":\"3500\",\"WeaTempStopCH2\":\"3500\","
[21:55:43][D][uart_debug:158]: <<< "\"WeaCorr\":\"0\",\"WeaObjCorr\":\"NN\",\"BuTempAct\":\"3223\",\"BuTotalFuel\":\"649475\",\"Bu24hFuel\":\"4924\","
[21:55:43][D][uart_debug:158]: <<< "\"BuActualFuel\":\"0\",\"BuModulMin\":\"15\",\"BuModulMax\":\"100\",\"BuTimeAboveMax\":\"10\",\"B026\":\"5\","
[21:55:43][D][uart_debug:158]: <<< "\"BuAirMin\":\"14\",\"BuAirMax\":\"100\",\"BuAditAir\":\"10\",\"BuFuelCaloric\":\"69\",\"BuFuelCorr\":\"-4\",\"BuMode\":\"Auto\",\"BuSbyPeriod\":\"20\",\"BuSbyAirTime\":\"15\",\"BuSby"
[21:55:43][D][uart_debug:158]: <<< "AirPwr\":\"50\",\"BuSbyFeedTime\":\"10\","
[21:55:43][D][uart_debug:158]: <<< "\"BuIntFeedTime\":\"60\",\"BuIntBreakTime\":\"500\",\"BuIntAirPwr\":\"25\",\"BuIntHist\":\"500\",\"BuOptNoFireTime\":\"120\",\"BuOptPwr\":\"39\",\"BuOptFdrEff\":\"1680\",\"BuOptFd"
[21:55:43][D][uart_debug:158]: <<< "rFTime\":\"11\",\"BuOptClrPeriod\":\"20\",\"BuOptClrTime\":\"15\","
[21:55:43][D][uart_debug:158]: <<< "\"B062\":\"1395\",\"B060\":\"150\",\"B059\":\"20\",\"DHWDays\":\"0010100000000000000000000000\",\"DHWHours\":\"000000100000000000000000000000000000000000000000\","
[21:55:43][D][uart_debug:158]: <<< "\"DHWCDays\":\"0000000000000000000000000000\",\"DHWCHours\":\"000000000000000000000000000000000000000000000000\",\"CH1RDays\":\"0000000000000000000000000000\",\"CH"
[21:55:44][D][uart_debug:158]: <<< "1RHours\":\"000000000000000000000000000000000000000000000000\",\"CH2Days\":\"0000000000000000000000000000\",\"CH2Hours\":\"0000000000000000000000000000000000000"
[21:55:44][D][uart_debug:158]: <<< "00000000000\",\"DHWCMode\":\"Stop\",\"DHWCAlwaysON\":\"On\",\"DHWCTempON\":\"4000\",\"DHWCWork\":\"5\","
[21:55:44][D][uart_debug:158]: <<< "\"DHWCBrake\":\"0\",\"DevType\":\"SKZP-02S V2.46 2021-04-21-ST\",\"K003\":\"5000\",\"K004\":\"1111111000000000000000000000\",\"K005\":\"111111111111111111111111111111111"
[21:55:44][D][uart_debug:158]: <<< "111111111111111\","
[21:55:44][D][text_sensor:067]: 'uart_readline': Sending state '{"DevId":"TASI7 S8","DevPin":"TASI7 OKN","Token":"GQMMW","FrameType":"SkzpData","TimeStamp":"22-05-10","BoilerTempAct":"4542","AN01":"1420","CH1ReturnTempAct":"3718","CH1ReturnTem'
[21:55:44][D][uart_debug:158]: <<< "\"K006\":\"1\",\"K007\":

sargus123, 2022-01-03:
DevType\":\"SKZP-02S V2.46 2021-04-21 ST\",\"K003\":\"5000\",
\"K004\":\"1111111000000000000000000000\",
\"K005\":\"111111111111111111111111111111111" 
 '{"DevId":"TASI7 S8","DevPin":"TASI7 OKN","Token":"GDMHD","FrameType":"SkzpData"' 
"111111111111111\",\"K006\":\"1\",\"K007\":\"0\"}\r\n" 
"{\"DevId\":\"TASI7 S8\",
\"DevPin\":\"TASI7 OKN\",
\"Token\":\"GDMHD\",
\"FrameType\":\"SkzpData\",
\"TimeStamp\":\"19-35-33\","  "\"BoilerTempAct\":\"4921\",
\"AN01\":\"1120\",
\"CH1ReturnTempAct\":\"3968\",
\"CH1ReturnTempCmd\":\"3000\",
\"CH1ReturnProtAct\":\"On\",
\"BoilerTempCmd\":\"5000\",
\"DevStatus\":\"M"  "OD03804200001\",
\"Alarms\":\"00000000I0K00000000000\",
\"DHWTempAct\":\"3099\",
\"DHWTempCmd\":\"7800\",
\"DHWHist\":\"500\",
\"DHWMode\":\"Stop\",
\"DHWPriority\":\"Off\",
\"DHWOverH\":\"1000\",
\"UpTime\":\"0d1h43m16s\",
\"CH1MixTempAct\":\"30000\",
\"CH1MixTempCmd\":\"0\",
\"CH1MixTempBase\":\"7500\",
\"CH1MixTempMin\":\"2500\",
\"CH1MixTempMax\":\"8000\",
\"C046\":\"0\",
\"CH1MixValueAct\":\"0\",
\"CH1MixValueMin\":\"10\",
\"CH1MixValueMax\":\"100\",
\"CH1MixGain\":\"10\",
\"CH1MixPeriod\":\"20\",

McMabb, 2026-10-25:
Hi, in case it helps anyone still looking at this: I got the SKZP-02 (SKZP-02S V2.46) fully readable in ESPHome over its DB9 COM port.

The controller pushes a JSON SkzpData frame by itself every ~3 s at 57600 8N1, so no polling and no Modbus are needed.
Use an RS232-TTL module (MAX3232/SP3232) with only TX→ wired to the ESP32 and no tx_pin. Driving the line from the ESP32 blocks the data and can even reset the controller.
DevStatus decodes to state (OFF/STO/ROZ/MOD), burner modulation %, fan %, feeder and four pumps (CO1, CO2, DHW, DHW circulation). B062 is the fuel left in the hopper (×10 kg).

Config, wiring and the field map are here: https://github.com/McMabb/timel-skzp-esphome

It’s read-only so far. If anyone knows how the controller accepts setpoint changes, I’d be glad to add it.

piaassek, 2026-10-26:
If you want to edit the values, here is an example of the code I use.
In my case, DevId and DevPin can be random; the important thing is that the number of characters matches.

~ echo '{"DevId":"GSDFGH","DevPin":"98003","Token":"DFEDGE","CH1RoomTempCom":"2300"}' | nc 192.168.1.157 1515