esphome/feature-requests #1580: include ADS1015 adc (compatible with ADS1115 but only 12 bit)

democles9308, 2022-01-06:
**Describe the problem you have/What new integration you would like**
Integrate the usage of an ADS1015, the 12bit variant of the ADS1115. Only some minor changes  to do the conversion 14>12 bits...

**Please describe your use case for this integration and alternatives you've tried:**
I used the ADS1115 component  for an ADS1015 and it seems to work, correct readings

ads1115:
  - address: 0x49 ( the address of my ADS1015)  
    continuous_mode: true
sensor:
  - platform: ads1115
    multiplexer: 'A0_GND'
    gain: 4.096
    accuracy_decimals: 3
    update_interval: 1s
    name: "ADS1015 A0-GND"
    internal: true
    id: adc0

**Additional context**
Is there no modification needed to take the difference of resolution (12bits only)  into account ?
Thxs.

martgras, 2022-01-09:
No there is no modification required to use ADS1015 with the `ads1115`  component.    From a developer point of view there is no difference interfacing between ADS1015 and ADS1115

martgras, 2022-01-18:
Closings since ads1015 is already supported

closed 2022-01-18 by martgras

haxney, 2022-02-22:
I stumbled across this issue while asking the same question "can ESPHome support the ADS1015". The only request I'd have is to add an explicit mention to the documentation page that the `ads1115` component works for the ADS1015 without any changes.