esphome/feature-requests #1581: Add support for max_value/min_value

spacemanspiff2007, 2022-01-07:
**Describe the problem you have/What new integration you would like**

Ability to limit the value of a sensor to a min/max value

**Please describe your use case for this integration and alternatives you've tried:**

```
  - lambda: |-
      if (x >= 150.0) { return 150.0;}
      return x
```

**Additional context**

I know it's possible with a lambda but it feels like this should be supported out of the box with ``max_value`` and ``min_value`` in the filters.

sorattorafa, 2022-06-27:
Please, can u assign this issue to @euEmica?

markusg1234, 2022-07-01:
I like this, seems to me has been missing. 
One example I would use it for is a moisture capacitance sensor using an ADC that some times decides to display 101-103%. Would be good to just set a max of 100% using a filter.

closed 2023-07-30 by jesserockz