esphome/feature-requests #1582: Add support for Waveshare 2.13 LOLIN e-paper (212 x 104)

Tokn59, 2022-01-07:
**Describe the problem you have/What new integration you would like**
Add support for this Waveshare e-paper module. I believe it uses library #include <GxGDEW0213Z16/GxGDEW0213Z16.h>  // 2.13" b/w/r from GxEPD lib.

**Please describe your use case for this integration and alternatives you've tried:**
Use this display in ESPHome. I have tried all currently supported models but none of them work:

#    model: 2.13in-ttgo
#    model: 2.13in
#    model: 2.13in-ttgo-b73
#    model: 2.13in-ttgo-b74 
#    model: 2.13in-ttgo-b1 
#    model: 2.13in-ttgo-dke  

**Additional context**
The display is working fine with other software.
If it should work with current supported models please let me know.
Willing/ready to Beta test.

EraYaN, 2022-07-22:
And this probably also goes for this one: https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT_(B)_Manual#Introduction As the actual Waveshare version.

KinDR007, 2022-11-25:
here is [source code](https://github.com/wemos/LOLIN_EPD_Library/tree/master/src)for UC8151D please add to esp home