esphome/feature-requests #1586: Inkbird IBS-TH2

bschatzow, 2022-01-08:
Esphome has support for the Inkbird IBS-TH1 but not the newer IBS-TH2.

**Describe the problem you have/What new integration you would like**

Support for this ble sensor. People that tried using the Inkbird IBS-TH1 platform get wrong readings below zero.
**Please describe your use case for this integration and alternatives you've tried:**
Need support for the new sensor.

**Additional context**

srnoth, 2022-01-09:
Seems it may do the same on the TH1s:

https://community.home-assistant.io/t/inkbird-esp32-displaying-wrong-temp-when-negative/327920

OscarHanzely, 2022-01-30:
This would be awesome if ESPHome support this sensor. Many other projects like Blerry or BLE Monitor that goes via MQTT already support it, so fingers crossed it will be soon available natively in ESPHome :)

dustin-git, 2022-02-21:
another request for IBS-TH2 support. Please!

efaden, 2022-03-17:
Blerry has a driver... https://github.com/tony-fav/tasmota-blerry/blob/main/blerry/drivers/blerry_driver_IBSTH2.be. My cpp is a little rusty (20 years or so).  But I'll look at converting later.

han9988, 2022-04-04:
I would like to kindly request support for IBS-TH2, too. 
I confused "Passive BLE Monitor" with "ESPHome BLE Tracker" when I bought the device.
In "Passive BLE Monitor" the device is supported: https://custom-components.github.io/ble_monitor/by_brand#inkbird 
Thought i could use the "inkbird_ibsth1_mini" platform and get away with it but apparently not.
I'm willing to help in any way needed, for example to try things out, provide more information, etc.

[12:49:38][VV][esp32_ble_tracker:497]: Parse Result:
[12:49:38][VV][esp32_ble_tracker:514]:   Address: 49:42:08:00:45:FE (PUBLIC)
[12:49:38][VV][esp32_ble_tracker:516]:   RSSI: -82
[12:49:38][VV][esp32_ble_tracker:517]:   Name: 'sps'
[12:49:38][VV][esp32_ble_tracker:525]:   Ad Flag: 6
[12:49:38][VV][esp32_ble_tracker:528]:   Service UUID: 0xFFF0
[12:49:38][VV][esp32_ble_tracker:531]:   Manufacturer data: AB.10.00.D7.8B.64.06 (7)
[12:49:38][VV][esp32_ble_tracker:547]: Adv data: 02.01.06.03.02.F0.FF.04.09.73.70.73.0A.FF.75.08.AB.10.00.D7.8B.64.06 (23)
[12:49:38][VV][xiaomi_lywsd03mmc:021]: parse_device(): unknown MAC address.
[12:49:38][VV][xiaomi_lywsd03mmc:021]: parse_device(): unknown MAC address.
[12:49:38][VV][xiaomi_mhoc401:021]: parse_device(): unknown MAC address.
[12:49:38][VV][inkbird_ibsth1_mini:059]: parse_device(): unexpected data
[12:49:38][D][esp32_ble_tracker:723]: Found device 49:42:08:00:45:FE RSSI=-82
[12:49:38][D][esp32_ble_tracker:744]:   Address Type: PUBLIC
[12:49:38][D][esp32_ble_tracker:746]:   Name: 'sps'
[12:49:38][V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.10 s).
[12:49:39][V][component:200]: Components should block for at most 20-30ms.

ciband, 2022-04-21:
I would like support for this as well.  Thank you.

MACscr, 2022-07-12:
I am having no luck with it. Even weird is that im not getting a name for my IBS-TH2 in the ble_tracker output. I have double checked the mac address though against the mac listed in the Engbird ios app though and its working fine there.

Yonghui, 2022-07-27:
I think that support for IBS-TH2 can be done by simply removing the following mnf_data.data[6] !=8 check.
 
esphome/components/inkbird_ibsth1_mini/inkbird_ibsth1_mini.cpp

Line 58:
  if (mnf_data.data[6] != 8) {
    ESP_LOGVV(TAG, "parse_device(): unexpected data");
    return false;
  }

As @han9988 comment indicated, for IBS-TH2 or  IBS-TH2 plus device, mnf_data.data[6] value is "06"

[12:49:38][VV][esp32_ble_tracker:528]: Service UUID: 0xFFF0
[12:49:38][VV][esp32_ble_tracker:531]: Manufacturer data: AB.10.00.D7.8B.64.06 (7)
[12:49:38][VV][inkbird_ibsth1_mini:059]: parse_device(): unexpected data

And I think mnf_data.data[6] != 8 check is not necessary.

Yonghui, 2022-07-30:
I successfully get the IBS-TH2 plus data into esphome by forking 
https://github.com/esphome/esphome 
into for example:
https://github.com/YOURFORK/esphome/

Then remove the 4 lines starting from Line 58, esphome/components/inkbird_ibsth1_mini/inkbird_ibsth1_mini.cpp

Use your modified inkbird_ibsth1_mini component with the following config

external_components:
  - source:
      type: git
      url: https://github.com/YOURFORK/esphome/
      ref: dev
    components: [ inkbird_ibsth1_mini ]

closed 2022-08-01 by jesserockz