esphome/feature-requests #1589: Add support for BL0930

eduperez, 2022-01-09:
**Describe the problem you have/What new integration you would like**

Please, add support for BL0930 energy meter.

**Please describe your use case for this integration and alternatives you've tried:**

Used in Sonoff Dual R3 **v2**.

**Additional context**

eduperez, 2022-01-19:
Perhaps it is compatible with the [BL0940](https://esphome.io/components/sensor/bl0940.html) sensor?

eduperez, 2022-01-20:
I'll answer myself: unless there is some special configuration needed... no, it is not compatible.

gabioancea, 2022-01-24:
BL0940 is not compatible with BL0930 :)

Dacesilian, 2022-02-04:
I would like to have this support for Sonoff Dual R3 v2 metering as well!: )

closed 2022-04-16 by eduperez

poldim, 2022-05-15:
@eduperez what was the resolution to this that you closed it? Does the BL0939 also work for the BL0930?

eduperez, 2022-05-16:
@poldim Yes, the code at #1630 works perfectly with these devices.

andrewigali, 2022-07-21:
> @poldim Yes, the code at #1630 works perfectly with these devices.

I have some Sonoff Dual R3 v2.0 devices.
I tried with bl0939 but it was not fully installed.
Please share your total ESPHome FW code.
This is my code (from sensor:):
sensor:
  - platform: uptime
    name: Spájz világítás és konnektor üzemidő
    filters:
      - lambda: return x / 3600;
    unit_of_measurement: "óra"
    accuracy_decimals: 2
    update_interval: 1800s

  - platform: wifi_signal
    name: Spájz világítás és konnektor WiFi jelerősség
    update_interval: 1800s

  - platform: bl0939 # Sonoff Dual R3 1.6-nál cse7761 kell. Sonoff Dual R3 2.0-nál bl0939 kell.
    update_interval: 60s
    voltage:
      name: Spájz világítás és konnektor feszültség # "$long_devicename Voltage"
    current_1:
      name: Spájz konnektor áramerősség # "$long_devicename Current 1"
    current_2:
      name: Spájz világítás áramerősség # "$long_devicename Current 2"
    active_power_1:
      name: Spájz konnektor teljesítmény # "$long_devicename Power 1"
      unit_of_measurement: W
      id: "spajz_konnektor_wattage"
      icon: mdi:lightning-bolt-circle
    active_power_2:
      name: Spájz világítás teljesítmény # "$long_devicename Power 2"
      unit_of_measurement: W
      id: "spajz_vilagitas_wattage"
      icon: mdi:lightning-bolt-circle

  - platform: total_daily_energy
    name: "Spájz konnektor fogyasztás"
    power_id: "spajz_konnektor_wattage"
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001 # Ezt ne módosítsd.
    unit_of_measurement: kWh
    accuracy_decimals: 5 # A kapott értéknél a tizedesjegy utáni számok mennyisége.
    icon: mdi:clock-alert

  - platform: total_daily_energy
    name: "Spájz világítás fogyasztás"
    power_id: "spajz_vilagitas_wattage"
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001 # Ezt ne módosítsd.
    unit_of_measurement: kWh
    accuracy_decimals: 5 # A kapott értéknél a tizedesjegy utáni számok mennyisége.
    icon: mdi:clock-alert

No measure results. :(
Please help me!

eduperez, 2022-07-24:
My code is much simpler:

```
[eperez@delfin ESPHome]$ cat sonoff1.yaml 
substitutions:
  name:   Sonoff 1
  device: sonoff1

packages:  
  device: !include include/esp32dev.yaml
  common: !include include/common.yaml
  extras: !include include/sonoff_dr3.yaml

api:
```

```
[eperez@delfin ESPHome]$ cat include/esp32dev.yaml 
esphome:
  name: ${device}
  platform: ESP32
  board: esp32dev
```

```
[eperez@delfin ESPHome]$ cat include/common.yaml 
wifi:
  ssid: "[REDACTED]"
  password: "[REDACTED]"
  domain: "[REDACTED]"

ota:

logger:
  baud_rate: 0

sensor:
  - platform: uptime
    id: ${device}_uptime
    name: "${name} Uptime"
  - platform: wifi_signal
    id: ${device}_wifi_signal
    name: "${name} Signal Sensor"
    update_interval: 60s
```

```
[eperez@delfin ESPHome]$ cat include/sonoff_dr3.yaml 
uart:
  id: uart_bus
  tx_pin: GPIO25
  rx_pin: GPIO26
  baud_rate: 4800
  parity:    NONE
  stop_bits: 1

output:
  - platform: gpio
    pin: GPIO27
    id: relay1
  - platform: gpio
    pin: GPIO14
    id: relay2

switch:
  - platform: output
    name: "$name - relay 1"
    output: relay1
    id: switch1
  - platform: output
    name: "$name - relay 2"
    output: relay2
    id: switch2

status_led:
  pin:
    number: GPIO13
    inverted: yes

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "$name button"
    on_press:
      - logger.log: "$name button"
  - platform: gpio
    pin:
      number: GPIO32
      mode: INPUT_PULLUP
      inverted: True
    name: "$name switch 1"
    on_state:
      - logger.log: "$name switch 1"
      - switch.toggle: switch1
  - platform: gpio
    pin:
      number: GPIO33
      mode: INPUT_PULLUP
      inverted: True
    name: "$name switch 2"
    on_state:
      - logger.log: "$name switch 2"
      - switch.toggle: switch2

sensor:
  - platform: bl0939
    update_interval: 60s
    voltage:
      name: '$name Voltage'
    current_1:
      name: '$name Current 1'
    current_2:
      name: '$name Current 2'
    active_power_1:
      name: '$name Active Power 1'
    active_power_2:
      name: '$name Active Power 2'
    energy_1:
      name: '$name Energy 1'
    energy_2:
      name: '$name Energy 2'
```

Hope this helps!

justaskumelis, 2022-08-26:
You are mixing BL0930 with BL0939 and they are not the same. BL0939 communicates thru uart bus, BL0930 has a low frequency output (https://www.belling.com.cn/media/file_object/bel_product/BL0930E/datasheet/BL0930E_V1.0_en.pdf) So there is no support for it so far?

eduperez, 2022-08-26:
> You are mixing BL0930 with BL0939 and they are not the same. BL0939 communicates thru uart bus, BL0930 has a low frequency output (https://www.belling.com.cn/media/file_object/bel_product/BL0930E/datasheet/BL0930E_V1.0_en.pdf) So there is no support for it so far?

Well, I probably copied the wrong chip name from some random website... However this issue was about the power meter used in Sonoff Dual R3 v2 devices, and it was already added.