esphome/feature-requests #1591: Mobile Data Connection

tobru, 2022-01-10:
**Describe the problem you have/What new integration you would like**

I'd like to add a mobile data connection to my ESPHome. For example:

* M5Stack COM.NB-IoT Modul SIM7020G
* M5Stack COM.LTE-DATA Modul A7600C1

These are only examples, but SIM7020G and A7600C1 seem to be widely used.

**Please describe your use case for this integration and alternatives you've tried:**

A use-case is to deploy an ESPHome device in a remote location (e.g. a camper) which only has mobile coverage (no Wifi).

**Additional context**

* https://docs.m5stack.com/en/module/comx_lte-data
* https://docs.m5stack.com/en/module/comx_nb-iot

nagyrobi, 2022-01-21:
These will be very popular after 5G SA (Standalone Network) will be deployed worldwide.

basilyohannan, 2022-02-20:
you could use your old phone to host a portable hotspot

PablexXXXX, 2023-09-29:
Oh man off course we need this thing. Imagine a esphome node with GPS and LTE using the wireguard component. PLEASE DO THIS. The ultimate portable esphome node. It would be ESPWORLD

PablexXXXX, 2023-09-29:
Ttgo T-sim7000g Module Esp32-wrover-b I know it is a sim7000g but you get the idea.

PablexXXXX, 2023-09-29:
> you could use your old phone to host a portable hotspot

That is not proffesinal bro.

JirikP, 2023-12-10:
Oh my god, yes. NB-IoT is something i have my eyes for long. GPS tracker or off-site device with direct line via wireguard to self hosted Home Assistant? Take my money.

JoHo94, 2024-01-18:
I +1 for this. Would be awesome! Currently building a smart camper van with esphome by sending the data via a portable wifi / lte router to home assistant. Energy is limited, so it would be a great advantage if I didn't need the router.

vahaldor, 2025-01-26:
I started working on it to track my bicycle. Have several vectors on how to achieve it. However, there are already devices on the market. Take [waveshare](https://www.waveshare.com/esp32-s3-a7670e-4g.htm) from China or much more sympathetic [Walter](https://www.tinytronics.nl/nl/development-boards/microcontroller-boards/met-telecommunicatie/walter-esp32-s3-iot-development-board-cat-m-nb-iot-en-gnss-gm02sp) that focus strictly on low energy consumption, NB-IOT etc. So the board is actually here. I haven't been successful in configuring Walter or Waveshare yet, just ESP32 as that is quite easier for a newbie to ESP as me.

I also plan to build with Rpi but it seems to be unnecessary. However, the biggest enemy is the power of course. 4G dongle still takes multiple times more than the ESP32 and my bicycle power it only when on battery because the bicycle disconnect everything when the main battery is charging, so it will always take some insignificant energy and kilometers from your everyday commute.

Technerd-SG, 2025-05-10:
Esphome needs LTE please!