esphome/feature-requests #1592: Implement an outlier sensor filter

tomlut, 2022-01-10:
**Describe the problem you have/What new integration you would like**

It would be advantageous if an outlier sensor filter could be implemented in ESPHome. Along the lines of the logic described here: https://www.home-assistant.io/integrations/filter/#outlier It is used to reject new sensor values that fall outside a specified range from the mean of the last few readings (handy for rejecting spikes of noise).

**Please describe your use case for this integration and alternatives you've tried:**

At the moment this must be implemented in Home Assistant, necessitating an additional entity. Implementation of this simple filter in ESPHome would be more efficient.

**Additional context**

nagyrobi, 2022-07-08:
How about `quantile` or `median`?
https://esphome.io/components/sensor/index.html#quantile

tomlut, 2022-07-08:
Yep that'll do. Thanks!

closed 2022-07-08 by tomlut