esphome/feature-requests #1593: Give BMP280/BME280 configurable standby feature to improve accuracy of temperature measurement

DavyLandman, 2022-01-11:
**Describe the problem you have/What new integration you would like**
Improve the standby flag of the BMP/BME280 sensor to reduce internal overheating. The current 0.5ms is the setting suggested for gaming or indoor navigation, where you care more about delta's than accurate values. (see section 3.5)

I propose ESPHome either calculates an optimal standby period based on the `update_interval` and the sampling frequency, or gives the user an extra optional property to increase the standby period.

**Please describe your use case for this integration and alternatives you've tried:**

The BMP280/BME280 are quite nice sensors for measuring humidity/temperature/pressure. However the sensor is quite sensitive to external heat sources, if you manage to isolate that, they'll still measure a bit higher than for example a ds18b20 next to it. 

The documentation advice is to use the sensor standby mode to let it cool itself a bit. The [current code has hardcoded this to 0.5ms](https://github.com/esphome/esphome/blob/27364ee72c553f59aeb0743a033949b2e9d43cef/esphome/components/bme280/bme280.cpp#L135).  While even with the default measurement oversampling and update interval, a 1000ms standby would collect enough samples.

**Additional context**

Table of standby timings
[image]

[AdaFruit library also has this functionality](
https://github.com/adafruit/Adafruit_BMP280_Library/blob/486e1a802091794532b5ca0ec87ba15a2a8b31e8/Adafruit_BMP280.h#L163-L180)

DavyLandman, 2022-01-11:
Note: I'm not sure if we should expose this to the user, maybe I should raise an issue to change the default 0.5ms to a more reasonable standby time?

nagyrobi, 2022-01-12:
Default update interval is 60s. I think 1000ms is more than enough.

ohumi, 2022-04-19:
Either expose it to the user or increase default to higher value - I have to use deep sleep mode to cool it down right now. Otherwise it can show temperature up to 1deg higher (when in enclosure)

nagyrobi, 2022-04-20:
Or use a different sensor (like SHTC3) which doesn't exhibit this problem...

closed 2022-09-18 by jesserockz

amagr0, 2022-09-21:
It was implemented a fixed value or the user can change it?
I did not see any new configuration variable in [ESPhome BME280](https://esphome.io/components/sensor/bme280.html) page.

DavyLandman, 2022-09-21:
Use the force, read the source ;) https://github.com/esphome/esphome/pull/3804/files

It was changed to 1000ms, which I think is a much better default.