esphome/feature-requests #1594: Switch to number component for PID control parameters

Tuckie, 2022-01-11:
**Describe the problem you have/What new integration you would like**

The control parameters should utilize the number component for easy viewing and tuning within home assistant.

**Please describe your use case for this integration and alternatives you've tried:**

Currently, the autotune process requires back and forth with multiple flashes, log viewing, and custom yaml for each environment. Implementing this would allow for a much easier tuning experience.

**Additional context**

Weissnix4711, 2022-12-03:
Just a heads-up this can already be achieved with template number components and the `climate.pid.set_control_parameters` action.

```yaml
number:
  - platform: template
    name: "${devicename} Room Influence Proportional"
    id: thermostat_influence_kp
    mode: box
    entity_category: config
    min_value: 0
    max_value: 100
    step: 0.0001
    lambda: |-
      return id(room_influence).get_kp();
    set_action:
      - climate.pid.set_control_parameters:
          id: room_influence
          kp: !lambda |-
            return x;
          ki: !lambda |-
            return id(thermostat_influence_ki).state;
  - platform: template
    name: "${devicename} Room Influence Integral"
    id: thermostat_influence_ki
    mode: box
    entity_category: config
    min_value: 0
    max_value: 100
    step: 0.0001
    lambda: |-
      return id(room_influence).get_ki();
    set_action:
      - climate.pid.set_control_parameters:
          id: room_influence
          kp: !lambda |-
            return id(thermostat_influence_kp).state;
          ki: !lambda |-
            return x;
```

It's long, and sorta ugly, but it works just fine.

nagyrobi, 2022-12-03:
`room_influence` is the ID of your PID climate?

roma1141r1, 2024-09-25:
Thanks! My friend!