esphome/feature-requests #1595: DeepSleep Wisualization on dashboard

GamerClassN7, 2022-01-12:
**Describe the problem you have/What new integration you would like**
Visual Change to ESP HOme dashboard to show taht device is in deep sleep
[image]
for example yealov color and maybe time to next reconect ??
[image]

**Please describe your use case for this integration and alternatives you've tried:**
i have many deep slepp senzors and if they are all offline, it is not easlly noticable what are actually ofline or just sleeping :(

**Additional context**
DeepSleep/Native API

GamerClassN7, 2022-01-20:
can someone point me where SourceFIle of dashboard are stored i can try to implement it on my own :)

Valcob, 2022-01-30:
https://github.com/esphome/hassio/tree/main/esphome
here you go 
and have fun!

GamerClassN7, 2022-03-15:
> https://github.com/esphome/hassio/tree/main/esphome here you go and have fun!

Unforchunatelly, i tryed but i am unable to get orientet in TS code it is to aliane to me, i was mistakent taht it is writen in JS.

GamerClassN7, 2023-10-04:
Is there any possibility to integrate this feature ? in near feature ?

ssieb, 2023-10-04:
I don't see how this is even possible.  The dashboard has no way to know that the device is in deep sleep.

GamerClassN7, 2023-10-04:
hmm... Dashboard has acces to Conf file of each device right ? so if there is deep_sleep function used it know it device is in sleep if  time from last connect to now is bigger that deep_sleep interval. Tat was original idea behind this feature.

[image]

ssieb, 2023-10-04:
But there's no way to know if it's actually in deep sleep or just offline.  Also, the durations aren't necessarily set, they can be changed when deep sleep is triggered.  And the dashboard might not contact it while it's online.

GamerClassN7, 2023-10-05:
> But there's no way to know if it's actually in deep sleep or just offline. Also, the durations aren't necessarily set, they can be changed when deep sleep is triggered. And the dashboard might not contact it while it's online.

That is also true how ever on other hand devices are not offline but in deep sleep so would be nice to at least statically change color from red to yellow or other when offline and device has "deep_sleep" in configuration