esphome/feature-requests #1597: Organize nodes in gruop or zone

worldwidefelix, 2022-01-13:
**Describe the problem you have/What new integration you would like**
 Organize nodes in Group, zone etc with drop down menu.

**Please describe your use case for this integration and alternatives you've tried:**

I have dozens of ESP node in my esphome. Sometimes it is hard and complicated to find my node only with its name. So it would be great to put nodes to group(zone) to visualize clearly.

**Additional context**

mkaatman, 2022-01-16:
Maybe a filename filter at the top would be a quick solution?

worldwidefelix, 2022-01-16:
I would prefer group or zone function as I have multiple esp nodes at
several locations.

alainsch, 2022-11-25:
Maybe, the possibility to put scripts into subfolders under esphome/scripts folder and automatically create a dashboard per folder ?