esphome/feature-requests #1598: Add USC CDC client mode support and USB Logger

unaiur, 2022-01-13:
ESP32S2 and USP32S3 has integrated USB support and modules usually do not include a USB -> serial bridge. 
But EspHome does not support USB CDC, leaving no easy path forward. Some modules (like LOLIN S2 mini) not even expose the UART0 pins.

**Describe the problem you have/What new integration you would like**
Add support for USB CDC UART port in the logger component, to avoid using one of the hardware UARTs for this purpose.

**Please describe your use case for this integration and alternatives you've tried:**
- Alternative A: using a hardware UART. It requires extra hardware (an USB to serial bridge) and consumes one of the two UARTs in ESP32S2/C3.
- Alternative B: using the API to access logs; but it does not support all the use cases; for example, if a component fails and causes the MCU to reboot, the logs are lost.

**Additional context**

rbray89, 2022-01-20:
I'm interested in this, but for Host USB. I have a device that the only way to communicate with it is to connect it to a USB Host and treat it as a CDC ACM device.

Ideally the UART would be extended to make this as generic as possible.

unaiur, 2022-01-21:
That would not be handled by the logger.

You would need a full USB stack instead of the one that is integrated into
the chip ROM memory and can only handle logging and firmware upgrades.

Best regards!

closed 2022-05-10 by jesserockz