esphome/feature-requests #1600: Support for Adafruit QT Py ESP32-S2

ashald, 2022-01-15:
**Describe the problem you have/What new integration you would like**
I want to be able to use ESPHome with [Adafruit QT Py ESP32-S2](https://www.adafruit.com/product/5325).

**Please describe your use case for this integration and alternatives you've tried:**
It appears to be a legit ESP32-S2 board that is very small in size and has convenient Stemma QT port for chaining multiple sensors.

**Additional context**
I tried installing some sort of basic firmware that would just connect to the network in ways described in https://github.com/esphome/feature-requests/issues/1511 (namely, WebSerial in Chrome, and ESPHome WebSerial).

When I flash the board via Adafruit's WebSerial tool (need to press reset while holding boot), I can erase and write the bin file generated by ESPHome, but the device newer joins network, nor shows up in ESPHome flasher to check logs.

When I try to flash it with ESPHome WebSerial I get `Installation failed: Error: Configuration does not match the platform of the connected device. Expected a ESP32 device.`

niamor29, 2022-02-26:
I'm looking to do the same thing. No solution at the moment

zala143, 2022-02-27:
same here - great little board, would love to use with ESPHome

mantrapdx, 2022-04-15:
Mark me down as another person that would love to see this board supported.

grassystew, 2022-05-17:
I would also love to see support on this board.  
I have tried many different flashing attempts like OP but nothing ever works.  I can never see logs and looks like it is in a boot loop of some sort post flash. 
I followed the suggestions here but these didn't help either: https://github.com/esphome/issues/issues/2931

I am assuming there is something unique on this board not liked.

ashald, 2022-05-19:
Try this config - that's what worked for me (note it's esp-idf only, doesn't support Arudino):
```
esphome:
  name: qtpy-esp32s2

external_components:
  - source: github://unaiur/esphome@dev-stdout-logger
    components:
      - logger

esp32:
  board: featheresp32-s2
  variant: esp32s2
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP_CONSOLE_USB_CDC: y

logger:
  hardware_uart: stdout

api:
ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

i2c:
  - id: stemma
    sda: 41
    scl: 40
    scan: true

sensor:
...
```

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/996

closed 2022-07-01 by nagyrobi