esphome/feature-requests #1602: Support always available configuration entities

elupus, 2022-01-18:
**Describe the problem you have/What new integration you would like**

Ability to mark an entity as configuration, and make it available even if the device is itself is not available. They should be restore state entites in home assistant, and once a connection is established they should update their state in esphome with the current state of the entity in home assistant.

**Please describe your use case for this integration and alternatives you've tried:**

This allow configuration of things like deep sleep disablement over normal native home assistant connection. This should work similar to Home Assistant Sensor, but it should not require a helper configuration inside home assistant to be created beforehand. 

**Additional context**

nagyrobi, 2022-01-19:
Worth considering together with:
https://github.com/esphome/feature-requests/issues/1492
https://github.com/esphome/feature-requests/issues/1568

breti, 2022-04-18:
Not the same as, but related to #1686