esphome/feature-requests #1604: Update Selection - One Click Update

Blackshome, 2022-01-20:
**Describe the problem you have/What new integration you would like**
 

When updating it would be nice to be able to select the devices you would like (tick box) to be updated and click a (one) button and it will start the update of all selected devices. Just like the "update All" button but only the devices selected. This way it would be easer/faster to update the devices you would like and not the ones you don't want updated in one click. At the moment you have to update each one individually wait for the update then update the next one and so on. I am sure this would be very useful to everyone and would save a lot of time. It would also making it more user friendly.  

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

craigueh, 2022-01-21:
Yes, I agree this is an important one. I have over 50 ESP devices and clicking Update All often bombs out after maybe 10-15 if the tab refreshes or I go work on another window and come back. Then I have to start the Update All process again and it goes right back to the start and updates already "Up to date" devices. Even if there was an option to only update "Out of date" devices that would help a lot.

soloam, 2022-01-28:
I was going to open a feature request with this. This would be very welcome.

Blackshome, 2022-02-08:
Yea it would be so nice to have this selection update button. I am going through the updates now and it is painful, one at a time so slow.

Nathanc87, 2022-03-06:
I was actually about to request this but thought I would do a quick search. Would be extremely handy as I have some devices often offline (testing/templates) once it gets to these on the updates it stops because they are offline, then the manual process of updating starts.

alainsch, 2022-11-21:
If I do an update on my "shelly plug-s" shared script, I only want to update all plugs and not "update all" or do it plug-by-plug

iamvlanone, 2022-12-06:
I also have the issue of offline nodes and have requested some sort of folder/tab solution to move these devices into to get them off the main page. [#1995](https://github.com/esphome/feature-requests/issues/1995)

This will not help for the 50+ devices, for which the update selection would still be a great solution.

craigueh, 2022-12-06:
@iamvlanone Agreed, I also have about 20 additional offline devices that are there as templates like Sonoff Basic, Sonoff Mini, Shelly 1 etc. etc. I use these offline devices as a starting point for new devices but it's really annoying to have the dashboard cluttered with these offline devices the rest of the time. it would be great if there was a "convert to template" function and have the offline device move to a separate tab and off the main dashboard screen.

Additionally, a search function on the main screen would be a great help. Scrolling through almost 100 devices at this stage is pretty cumbersome, especially on a mobile device.
Not sure if anyone else has requested this yet, haven't seen it but I may have missed it.

bkaufx, 2022-12-07:
One workaround is to just move all config yaml files you don't want to compile into a temporary subfolder and then update all.