esphome/feature-requests #1605: Using native esp-idf components

dudanov, 2022-01-20:
**Describe the problem you have/What new integration you would like**

Option to use native esp-idf components in ESPHome project.

**Please describe your use case for this integration and alternatives you've tried:**

I need to use native esp-idf components, but there is no such possibility yet. It would be nice if there was an option like ``libraries`` that either copied the components specified in YAML to the project's ``components`` root directory, or used the ``EXTRA_COMPONENT_DIRS`` option in the root ``CMakeLists.txt`` file. These 2 ways are currently supported by ``Platformio``.

**Additional context**

dudanov, 2022-01-20:
Although in this case, the functionality of the ``external_components`` component is more suitable than the ``libraries`` option of the root component.

vidplace7, 2023-02-08:
https://github.com/esphome/esphome/pull/4000

mag1024, 2024-07-18:
> [esphome/esphome#4000](https://github.com/esphome/esphome/pull/4000)

This looks promising, but have you been able to get it to work?

I'm trying

```
esp32:
  variant: esp32s3
  board: lolin_s3_mini
  framework:
    type: esp-idf
    version: latest
    ...
    components:
      - name: esp_tinyusb 
        source: github://espressif/esp-usb
      - name: tinyusb
        source: github://espressif/tinyusb
```

and it doesn't appear to have any effect....

roger-, 2025-01-05:
Has anyone been able to get this working?