esphome/feature-requests #1608: GUI redesign - the elements are too large and such big cards don't bring value

homonto, 2022-01-22:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

dougle03, 2022-01-23:
+1 for this. Perhaps to a photoshop mockup of what you'd like to see instead. I agree though, I have over 50 entries and it is a bit busy. Could also benefit from a filter option to perhaps filter out offline, updates, etc...

homonto, 2022-01-23:
in fact is it not about "my liking" but simply small cards with small buttons only, maybe also as you say check box for update or so with the general button on top: "select/unselect all" - so in case I have 100 devices and I want to update only 98 I will "select all" and then "unselect my 2".
With this regard also "select only online" would be nice to update only the online ones

dougle03, 2022-01-23:
Fully agree. Would be great to implement a view option, that was either grid (Like it is now, but smaller boxes) or list with check boxes for update/delete and small icons for edit etc...

kkellner, 2022-12-08:
Would be nice if this was a table (with header filters) that just listed all my ESPHome devices.   One of the columns could be "Actions" that would allow buttons of Update, Visit, Edit, etc.   The ability to filter, sort and have a more compact list of all devices would be nice.    Also would be nice to have one of the columns be "board" (e.g., "esp32dev", "d1_mini") but I don't know if that info is available as metadata outside of the yaml.

fdcastel, 2023-06-18:
For large lists, cards are definitely not the way. 

A table approach would be much more useful. Unifi has some battle-tested designs which could serve as inspiration:

[image]