esphome/feature-requests #1610: Authentication for frontend

Kalypox, 2022-01-24:
**Authentication frontend**

It could be great to have possibility to protect this webpage :
[image]
My goal is to access this page from outside my local network, that's why a login/password could be great.

Thanks :)

randybb, 2022-01-24:
Like this? https://esphome.io/guides/cli.html?#dashboard-command

Kalypox, 2022-01-24:
Hello @randybb,
Yes it seems to be !
Thanks !

I'm using esphome as add-on in home assistant, do you know how can I add these parameters ?

randybb, 2022-01-24:
This is only for standalone or docker with some modification. I am not familiar with HASS plugins. Would stick with the default option - access it via HA exposed to the internet (web or app).

Kalypox, 2022-01-24:
Okay clear thanks !
I will try to use it with docker !

closed 2022-01-24 by Kalypox