esphome/feature-requests #1615: Add option to stop the integral term from updating when the climate is set to OFF

Guyohms, 2022-01-26:
**Describe the problem you have/What new integration you would like**
Actually, the integral term continues to evaluate even if the thermostat is OFF.
This will make the integral to saturate at `min_integral` or `max_integral`.
For a very slow system, it can take a while (10-12h) for the integral to find it's balance point.
I would see a new optional configuration variable like `keep_integral_when_off: true` (or something similar) so it's not a breaking change.

**Please describe your use case for this integration and alternatives you've tried:**
I am using the PID for a radiant heating floor which has a very long time constant. To meet my electricity provider's need, I lower my consuption at peak hours by putting the thermostat to OFF. When the peak period is over, I set it back to heat. At that point, the integral term is very high so it the temperature overshoots a good amount instead of just starting back from where it was and adjust as needed.

**Additional context**
Somewhat similar to https://github.com/esphome/feature-requests/issues/945

nagyrobi, 2022-01-27:
I'm in the same boat.

Guyohms, 2022-02-02:
As a workaround this can be done.
```
climate:
  - platform: pid
    id: plancher_chauffant
    ...
    on_state:
      - lambda: |-
          if (id(plancher_chauffant).mode == CLIMATE_MODE_OFF) {
            id(plancher_chauffant).set_max_integral(id(plancher_chauffant).get_integral_term());
            id(plancher_chauffant).set_min_integral(id(plancher_chauffant).get_integral_term());
          }
          else {
            id(plancher_chauffant).set_max_integral(id(pid_max_int).state);
            id(plancher_chauffant).set_min_integral(id(pid_min_int).state);
          }
```
It forces the integral to stay where it is while the climate is OFF and then restore the old values when it's not OFF.

`pid_min_int` and `pid_max_int` are `number` entities defined to set the integral value from the UI :  
```
number:
  - platform: template
    name:  ${devicefriendlyname} PID Min intégral
    unit_of_measurement: "%"
    min_value: 0.0
    max_value: 100.0
    step: 1.0
    mode: box
    set_action: 
      - lambda: id(plancher_chauffant).set_min_integral(x/100);
    optimistic: true
    restore_value: true
    initial_value: 0.0
    id: pid_min_int
```

sophof, 2023-06-14:
I have the same issue for a fan system that cools my receiver. Now that it is summer it will turn on full blast whenever my receiver switches on. In the meantime I have a workaround like this:
``` yaml
on_state:
  - if:
          condition:
            and:
              - lambda: 'return id(receiver_thermostat).mode != esphome::climate::ClimateMode::CLIMATE_MODE_OFF;'
              - switch.is_off: fan_supply
              - switch.is_off: pwm_on
          then:
            - climate.pid.reset_integral_term: receiver_thermostat # remove any integral term build up while off
            - switch.turn_on: fan_supply
            - switch.turn_on: pwm_on
            - logger.log: "Fan switched on"
```
Here the `fan_supply` powers my fans by relay and the `pwm_on` enables pwm control of the fans.