esphome/feature-requests #1619: SIM800L: Add FM radio support

nagyrobi, 2022-01-30:
**Describe the problem you have/What new integration you would like**
SIM800L: has a nice FM radio built-in. Has speaker output. Something like below would be nice:
- switch to turn radio ON/OFF (`AT+FMOPEN` and `AT+FMCLOSE`)
- number to set audio volume (`AT+FMVOLUME`)
- action to set FM frequency (`AT+FMFREQ`)
- action to start FM scanning (`AT+FMSCAN`)
- sensor with signal strength (`AT+FMSIGNAL`)

**Please describe your use case for this integration and alternatives you've tried:**
The module could be very useful during idle times when no GSM usage is active.

**Additional context**
N/A

nagyrobi, 2022-07-06:
This, combined with support for [I2S audio input](https://github.com/esphome/feature-requests/issues/1254) would make possible creation of FM-to-WebRadio receivers, which stream could be played in the network using other media players in any room.

CCChelios, 2022-11-22:
> This, combined with support for [I2S audio input](https://github.com/esphome/feature-requests/issues/1254) would make possible creation of FM-to-WebRadio receivers, which stream could be played in the network using other media players in any room.

I don't see the point in this. devices have access to the Internet and can play any audio stream.