esphome/feature-requests #1620: Add possibility to group states in webserver

spacemanspiff2007, 2022-02-02:
**Describe the problem you have/What new integration you would like**

I have multiple states and i would like to group them logically in the webserver. Currently the order is random.

**Please describe your use case for this integration and alternatives you've tried:**

If states can be grouped logically together then it's easier to keep an overview over the states

**Additional context**

Maybe an additional property on the item (e.g. ``group``) or an additional field on the webserver ``groups`` which is a list which contains lists of ids.

spacemanspiff2007, 2022-07-19:
I can't find any hint in the docs how grouping of sensors would be possible with v2.
Maybe I am missing something but I think it's not implemented.

randyoo, 2024-08-31:
> I can't find any hint in the docs how grouping of sensors would be possible with v2. Maybe I am missing something but I think it's not implemented.

It appears to be possible in v3, based on the screenshots in the docs, but I still don't see any mention of how to do it.