esphome/feature-requests #1621: HiTrend HT7017 energy monitor support

BSD31, 2022-02-03:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

[image]
[image]
[image]
[image]
[image]
[image]

[HT7017-CN.pdf](https://github.com/esphome/feature-requests/files/7993667/HT7017-CN.pdf)

jon-daemon, 2022-05-28:
I'd like to see that too. I have ordered the D52-2066 and I could provide data from it if needed.

BSD31, 2022-06-01:
I haven't made much progress on this subject for lack of time...
But I'm still very interested.
Have a good day