esphome/feature-requests #1623: OTA Safemode

mhetzi, 2022-02-04:
**Describe the problem you have/What new integration you would like**
Problem:
On Some of my devices, doing an OTA update without rebooting them in safemode is either slow or aborts.

Feature Request:
Adding a flag to the OTA component, so the Dashboard/CLI reboots the esp in safemode, waits for device, and then updates it.

**Please describe your use case for this integration and alternatives you've tried:**
Currently i restart them manually via HomeAssistant and then make the update in the Dashboard

**Additional context**

nagyrobi, 2022-07-19:
See also https://github.com/esphome/feature-requests/issues/1764

Ulrar, 2022-08-16:
Sometimes the linking can take so long that the device goes out of safe mode, meaning you have to try and 'time' it by enabling safe mode just before the linking finishes.
It's very tedious and I'd love to see that automated. I wonder if the on_begin hook could be used ?

nagyrobi, 2022-08-16:
See https://esphome.io/components/ota.html#configuration-variables
Increase your `reboot_timeout` as needed.