esphome/feature-requests #1624: Add support of distance TOF vl6180x

zigomatichub, 2022-02-04:
**Describe the problem you have/What new integration you would like**

Time of flight sensor can use VL6180 chip, it would be great to add this one in the actual list of TOF and distance sensor

**Please describe your use case for this integration and alternatives you've tried:**

There is this: https://github.com/pololu/vl6180x-arduino
Link to datasheet: https://www.st.com/resource/en/datasheet/vl6180.pdf

**Additional context**
Use I2C platform

wazoo666, 2022-03-13:
I also join to this request

please, could you create code to this short distance/light  ToF ??
thankU

I found 2 samples on github , but none is fully working

Raelx, 2022-12-28:
I too have a use for this.

ryan-durbin, 2023-08-13:
Following, would also like official support for VL6180

DocDoo, 2023-09-06:
Also following 👀 would love to have support for this combined time-of-flight distance and ambient light sensor ❤

medivb, 2024-01-26:
Definitely following this as well. The short range would be great to have and the custom component (besides being deprecated) is not working for me.

mikulik86, 2024-01-26:
I second that.

NonaSuomy, 2024-02-04:
> Definitely following this as well. The short range would be great to have and the custom component (besides being deprecated) is not working for me.

This is incorrect. This sensor is ACTIVE and not depreciated says right at the top of their pages. 

https://www.st.com/en/imaging-and-photonics-solutions/vl6180.html
https://www.st.com/en/imaging-and-photonics-solutions/vl6180x.html

> Also following 👀 would love to have support for this combined time-of-flight distance and ambient light sensor ❤

Just to note: Most breakout boards that say VL6180X on them are not X just VL6180's ie they have no ambient light sensor on them (I fell for this as I wanted the ambient light sensor as well but was sad when I got the module and tested it out).

medivb, 2024-02-04:
> > Definitely following this as well. The short range would be great to have and the custom component (besides being deprecated) is not working for me.
> 
> In case someone is confused about this comment, the hardware sensor is ACTIVE and not depreciated. The custom component that was made is being depreciated in the future [exxamalte/esphome-customisations#10](https://github.com/exxamalte/esphome-customisations/issues/10)
> 
> You can see the hardware sensor status on top of the product pages where it says Active on both:
> 
> https://www.st.com/en/imaging-and-photonics-solutions/vl6180.html https://www.st.com/en/imaging-and-photonics-solutions/vl6180x.html
> 
> > Also following 👀 would love to have support for this combined time-of-flight distance and ambient light sensor ❤
> 
> Just to note: Most breakout boards that say VL6180X on them are not X just VL6180's ie they have no ambient light sensor on them (I fell for this as I wanted the ambient light sensor as well but was sad when I got the module and tested it out).

I did not say the sensor was deprecated, but the custom component. It works, but a warning about this is shown on https://esphome.io/custom/custom_component.html.

NonaSuomy, 2024-02-05:
I just wanted to clarify. Didn't want someone who may quickly read that and wanted to tackle the code to be like why add this if the hardware is depreciated confusion.

NonaSuomy, 2024-05-30:
We just built a component for this but my sensor is a vl6180 (non x) anyone have a confirmed X version with the ALS (Ambient Light Sensor)?

wazoo666, 2024-05-31:
> We just built a component for this but my sensor is a vl6180 (non x) anyone have a confirmed X version with the ALS (Ambient Light Sensor)? [...]

Please, can you share the code ??

I am just too crazy, because some time ago I was able to get working vl6180 module under ESPhome, but my computer crashed and lost the code.
All the code I have tested does not compile (I have used the Pololu, DFRobot,... libraries).
I must admit that I am not a programming expert.
The custom sensor created in ESPhome has calls that are not recognised by the compiler.
I have tried to use AI but I get error in the folder location of the .h and .cpp files, and I am sure they are well pointed from the "external_component" in YAML.

If no sensor code added, compiler finish correctly, 
but my problem is when custom sensor and lambda code is used.
---------------------------------------------
 Sensor puerta
  - platform: custom
    id: dds238_cerradura
    lambda: |-
       auto my_sensor = new MyCustomSensor();
       App.register_component(my_sensor);
       return {my_sensor->dist_sensor, my_sensor->ambient_sensor};
    sensors:
      - name: "DDS238 cerradura"
        id: cierre_distancia
        unit_of_measurement: "mm"
        accuracy_decimals: 1

#Light sensor does not works. most Aliexpress modules are VL6180 no "L" version.  (return value=0)
      - name: "Ambient sensor"
        unit_of_measurement: "lux"
------------------------------------------------

NonaSuomy, 2024-05-31:
What are you using the sensor for?
Yes you can have it soon enough. I'm just trying to finish some simple hand gesture control.
Is your sensor the X version or V with no ALS?

AridRayne, 2024-06-01:
@NonaSuomy I have a vl6180x, with the ALS. I'd be more than happy to help out.

NonaSuomy, 2024-06-01:
Please contact me on ESPHome discord https://discord.gg/syUMbGXW same name.

wazoo666, 2024-06-03:
> Please contact me on ESPHome discord https://discord.gg/syUMbGXW same name.

OK, let me some time because need to recover my Discord account :D :D 

I am using a common Aliexp TOF050C module with IR filter cover, and it is vl6180 model w/o ligth sensor.  :(
I am using as a door lock detector.
I mean, on a security door it has diferent deep positioning, so I am reading and translating to ESPhome to generate different Automations in HA.
That is because a buy this model because I only need precise readings from 5mm to 60mm  :D

NonaSuomy, 2024-06-06:
@AridRayne can I get another ALS test if you have another free moment.

wazoo666, 2024-06-15:
> What are you using the sensor for? Yes you can have it soon enough. I'm just trying to finish some simple hand gesture control. Is your sensor the X version or V with no ALS?
> 

Finally I get it working.
I am not a great coding expert, just try and error methode :) 
I think my problem was includes and libraries typing sequence.
Now is working for my needs.
I only use distance measurement, my module does nos include light sensor.  It is basic vl6180  (w/o "X")

````yaml
esphome:
  name: pw-dds238
  friendly_name: pw-dds238
   includes:
      - ./custom_components/vl6180x_sensor/vl6180x_sensor.h
  libraries:
    - Wire
    - pololu/VL6180X  #@1.4.0
# -----------------------------------------------
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
# -----------------------------------------------
(logger, debug and API definitions)
.
.
# ----------------------------------------------
i2c:                            # COMUNICACION CON SENSOR I2C (VL6180) para ver cerradura de puerta
  id: i2c_bus
  scl: GPIO04
  sda: GPIO05
  frequency: 400kHz
  scan: true
# ---------------
.
.
.
# ---------------
sensor:
# Sensor puerta
  - platform: custom
    id: dds238_cerradura
    lambda: |-
      auto vl6180x_sensor = new esphome::vl6180x::VL6180XSensor();
      App.register_component(vl6180x_sensor);
      return {vl6180x_sensor};
    sensors:
      id: cerradura_distancia
      name: "DDS238 cerradura mm"
      unit_of_measurement: "mm"
      accuracy_decimals: 1
      icon: "mdi:arrow-expand-horizontal"
      filters:
        - multiply: 1.5
        - sliding_window_moving_average:
            window_size: 8
            send_every: 5
        - delta: 1.5
````

NonaSuomy, 2024-07-02:
```yaml
external_components:
#   - source:
#     type: local
#     path: components
  - source: github://NonaSuomy/esphome@vl6180x
    components: [ vl6180x ]
    refresh: 30s
 i2c:
  - id: i2cbus_001
    sda: 23
    scl: 22
sensor:
  - platform: vl6180x
    address: 0x29
    update_interval: 5s
    distance:
      id: vl6180x_distance
      name: "VL6180X Distance"
      unit_of_measurement: mm
    als:
      name: "VL6180X Ambient Light"
      unit_of_measurement: lux
      accuracy_decimals: 1
      gain: '2.5x'
      #under_glass: true
      #lux_without_glass: 500.0  # Add this line to specify the lux value without the glass
```

AridRayne, 2024-07-02:
Every now and then, I'd start getting a vcsel error after I changed my config. Completely removing power from the board and restarting fixed it.

NonaSuomy, 2024-07-02:
If you don't change the config is it stable after the power reset?

I no longer have a board to test it until I buy another one.

I have always had issue with I2C boards that keep the power always on even if the UC restarts and the only way to get them working again was to reset the power fully to them. I started building power switches into the power lines of the I2C boards. Maybe try using the enable pin on the board (shutx, etc) to a gpio pin.

AridRayne, 2024-07-02:
It only send to ever happen after flashing. But it's been stable after every power reset so far.

dmsergeevich63, 2024-07-25:
How to change the sensor address using the gpio pin? I need to use three sensors on one bus i2c?

I used the following settings with fast polling, the sensor works and does not heat up:
```
i2c:
  sda: GPIO21
  scl: GPIO22
  frequency: 400kHz

sensor:
  - platform: vl6180x
    address: 0x29
    update_interval: 0.1s
    distance:
      id: vl6180x_distance
      name: "VL6180X Distance"
      unit_of_measurement: mm
    als:
      name: "VL6180X Ambient Light"
      unit_of_measurement: lux
      accuracy_decimals: 1
      gain: '2.5x'
```

NonaSuomy, 2024-07-27:
Maybe check out TCA9548A 1-to-8 I2C multiplexers

https://forums.adafruit.com/viewtopic.php?t=119485

Their documentation for the other model says this:

```
The VL53L0X has a default I2C address of 0x29!

You can change it, but only in software. That means you have to wire the SHUTDOWN pin and hold all but one sensor in reset while you reconfigure one sensor at a time
```

ImNewty, 2024-07-30:
Using your code my sensor limits to 255mm on distance what could cause that?

NonaSuomy, 2024-07-30:
The range for the vl6180x is 5mm to 100mm depending on your light conditions with a MAXIMUM of 200mm in pristine condition not sure why you are expecting more than that from this unit. If you need more get the model up as it won't be very accurate otherwise.

ImNewty, 2024-07-30:
The sensor I got said it can measure up to 50cm so 500mm and also when used with the default arduino library it had no such issues.

NonaSuomy, 2024-07-31:
You probably used the pololu library, This is based on the adafruit library that doesn't make those claims. 255 feels like a limitation of the data type or a limitation on how I read the I2C data.
What are you using it for?

NonaSuomy, 2024-07-31:
Attempted a fix try now let me know. Sorry if it explodes.

srvoets, 2024-07-31:
I am trying to implement a hiletgo vl6180x, I am using the reccomended configs, I inirially was getting readings of 0 but now am getting "VCSEL Continuity Test; No measurement possible"

Board is a D1 Mini 8266

`i2c:
  - id: i2cbus_001
    sda: D2
    scl: D1
sensor:
  - platform: vl6180x
    address: 0x29
    update_interval: 10s
    distance:
      id: vl6180x_distance
      name: "Distance"
      unit_of_measurement: mm`

NonaSuomy, 2024-07-31:
There's a comment here about that not sure if the same error? https://github.com/esphome/feature-requests/issues/1624#issuecomment-2201941401
Did you try power cycling both the module and the esp after a successful upload (removing the power completely from both units)
I can't remember the VCEL error I was getting but it meant the sensor is dead.

NonaSuomy, 2024-07-31:
I get this error currently
```
[17:03:45][E][vl6180x.sensor:154]: Unknown error
[17:03:50][E][vl6180x.sensor:154]: Unknown error
[17:03:55][E][vl6180x.sensor:154]: Unknown error
[17:03:59][ 36945][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
```

So will have to see if it works for any one with a working module

srvoets, 2024-07-31:
> I get this error currently
> 
> ```
> [17:03:45][E][vl6180x.sensor:154]: Unknown error
> [17:03:50][E][vl6180x.sensor:154]: Unknown error
> [17:03:55][E][vl6180x.sensor:154]: Unknown error
> [17:08:54][154681][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
> [17:08:59][E][vl6180x.sensor:099]: System error; VCSEL Continuity Test; No measurement possible
> ```
> 
> So will have to see if it works for any one with a working module

I just wired up my last VL16180x, and am receiving this in logs, below is the relevant yaml

`[18:06:47][W][vl6180x:060]: Raw reading underflow
[18:06:47][D][sensor:093]: 'Distance': Sending state nan mm with 0 decimals of accuracy`

`esphome:
  name: tip-jar
  friendly_name: Tip-Jar
  includes:
    - "vl6180x_sensor.h"
  libraries:
    - SPI
    - Wire
    - adafruit/Adafruit BusIO @ 1.9.6
    - "Adafruit_VL6180X"
  

esp8266:
  board: d1_mini

sensor:
  - platform: custom
    lambda: |-
      auto vl6180x_sensor = new VL6180XSensor();
      App.register_component(vl6180x_sensor);
      return {vl6180x_sensor->distance_sensor};
    sensors:
      - name: "Distance"
        unit_of_measurement: mm
        accuracy_decimals: 0`

NonaSuomy, 2024-07-31:
Looks like it is a problem with my component code as I just tested with Arduino and my sensor I thought was dead is working fine note the Pololu library example is only showing up to 255 as well.

ImNewty, 2024-08-01:
With arduino I used the pololu library with scaling to get a longer range because I don't really care much about the resolution since I'm using this sensor so see how much food for animals I have left in an enclosed box.

NonaSuomy, 2024-08-01:
I wonder if this has to do with the vcsel need power cycle thing
https://github.com/adafruit/Adafruit_VL6180X/issues/15

ImNewty, 2024-08-02:
For me the error only happened when I didn't include the light sensor I only needed the distance and it went into that error but after adding light sensor as well it worked

ImNewty, 2024-08-20:
Any other updates on this?

k-korn, 2025-04-06:
Any chance to have this merged into ESPHome code? 
This [repo](https://github.com/NonaSuomy/esphome/tree/vl6180x/esphome/components/vl6180x ) works well for both VL6180 and VL6180x sensors.

Jezza34000, 2025-04-11:
+1 waiting for this

NonaSuomy, 2025-04-12:
Just to note the part is now obsolete 

[image]

[image]

Been trying to get the scaler working, currently testing it. It doesn't seem to actually give you any more distance just X whatever the distance by your scaler. Need more testing. However adding this feature required adding some code to track fresh out of boot which may or may not have fixed the vcel boot issue.

k-korn, 2025-04-12:
The device might be marked as obsolete, but no new device combines ALS sensor with ToF ranging - so it will still find use cases.

Also, a quick note on https://github.com/NonaSuomy/esphome/blob/vl6180x/esphome/components/vl6180x/vl6180x_sensor.cpp#L382 - ALS sensor measurement takes ~100ms, so if we read the value immediately, only the previous reading will be reported.

I've implemented proper async ALS and range measurement [here](https://github.com/k-korn/vl6180x-gesture-switch/blob/main/components/vl6180x_light/vl6180x_light.cpp#L302), feel free to take a look.

NonaSuomy, 2025-04-12:
Can you explain more technically how you were accomplishing gestures with the addition of ALS? I was attempting it before without the ALS which you can see commented out in the code (China said the one I bought was branded on the board vl6180x but it was only 0 no x.

k-korn, 2025-04-12:
On ALS giving 0 lx - you (and many of us) were fooled by the Chinese manufacturers. 
[This](https://i.imgur.com/5XolOeZ.png) is a VL6180 device, despite what the marking says. It behaves exactly as VL6180X would, but lacks the ALS sensor and reports zero values.

[This](https://i.imgur.com/CdgJkwS.png), on the other hand, is the proper VL6180X with the third optical window for the ALS sensor. 
It can be obvious if you compare the pictures on both datasheets, but many many chinese modules are labeled as VL6180X while not having the proper sensor.

As for the gestures, I am just re-implementing with VL6180X the devices I've had for almost 10 years now which use an IR LED + photodiode to detect user hand presence.

The implementation starts [here](https://github.com/k-korn/vl6180x-gesture-switch/blob/main/components/vl6180x_light/vl6180x_light.cpp#L428) and boils down to these items:

- We define min and max range where the device will be sensitive.
- If a hand is detected closer than the max range, [STATE_HAND](https://github.com/k-korn/vl6180x-gesture-switch/blob/main/components/vl6180x_light/vl6180x_light.cpp#L437) is entered.
- If the hand left the detection range in less that 1s, then the output is toggled (with optional acknowledgement of state change via aux LED)
- If the hand remains near, we enter brightness adjustment mode [STATE_ADJ](https://github.com/k-korn/vl6180x-gesture-switch/blob/main/components/vl6180x_light/vl6180x_light.cpp#L465). Output brightness is set in real time based on the received range, the closer the hand is the brighter the output.
- In 3 seconds the latest received brightness value becomes our new permanent state.
- IF the hand left in less than 3 seconds - adjustment mode is cancelled.

ALS sensor is not directly used, but it comes handy for a light control device anyway. Also VL6180X does have an impressive dynamic range, being able to detect light from 0.08 to 41,000 lx.

NonaSuomy, 2025-04-13:
Please have a test run. AGC added from k-korn and scaling is in as far as I can tell.

https://github.com/NonaSuomy/esphome/tree/vl6180x-pololu-scaler/esphome/components/vl6180x_pololu

```yaml
external_components:
#   - source:
#     type: local
#     path: components
  - source: github://NonaSuomy/esphome@vl6180x-pololu-scaler
    components: [ vl6180x_pololu ]
    refresh: 30s
i2c:
  - id: i2cbus_001
    sda: 23
    scl: 22
sensor:
  - platform: vl6180x
    address: 0x29
    update_interval: 5s
    distance:
      id: vl6180x_distance
      name: "VL6180X Distance"
      unit_of_measurement: mm
      scaling: 1 # 1, 2, or 3
    #als:
    #  name: "VL6180X Ambient Light"
    #  unit_of_measurement: lux
    #  accuracy_decimals: 1
```

You will need to remove the Gain config from your yaml as k-korn made it automatically adjust. Uncomment to use.

k-korn, 2025-04-13:
Tested the above code.

1. It tries to wait for ALS measurement synchronously and ESPHome is not happy about that:
```
[17:45:43][D][vl6180x:306]: ALS: 31.14 lx (Raw: 3893, Gain: 40.00x)
[17:45:43][W][component:237]: Component vl6180x.sensor took a long time for an operation (174 ms).
[17:45:43][W][component:238]: Components should block for at most 30 ms.
[17:45:45][D][vl6180x:249]: Distance: 274 mm
[17:45:45][D][vl6180x:306]: ALS: 31.18 lx (Raw: 3897, Gain: 40.00x)
[17:45:45][W][component:237]: Component vl6180x.sensor took a long time for an operation (174 ms).
[17:45:45][W][component:238]: Components should block for at most 30 ms.
[17:45:47][D][vl6180x:249]: Distance: 262 mm
[17:45:47][D][vl6180x:306]: ALS: 32.18 lx (Raw: 4022, Gain: 40.00x)
[17:45:47][W][component:237]: Component vl6180x.sensor took a long time for an operation (167 ms).
[17:45:47][W][component:238]: Components should block for at most 30 ms.

```
2. Scaler works.
3. ALS auto-gain works. 
4. I think folder name (components/vl6180x_pololu) should match component name, but that is fixed by copying folder content to local components.

So I think the implementation itself is working, but as ALS measurements take a lot of time - it would be better to implement a basic state machine in loop() and wait for ALS to be done asynchronously. I've implemented it [here](https://github.com/k-korn/vl6180x-gesture-switch/blob/main/components/vl6180x_light/vl6180x_light.cpp#L300) (other than that I request range measurements every time when ALS measurement process is not running)

NonaSuomy, 2025-04-13:
Knocked off the two low hanging fruit:
Changed the ESP_LOGV and folder name.

KoalaBear84, 2025-06-07:
As I also didn't check on esphome compatibility before I've bought this sensor and tested the above code.

All works good up to around 180mm. Then I tried the scaler, and it worked to ~360mm, but that is only because it looks it's just a filters: multiply: 2? And doesn't increase anything. Then tried to increase `update_interval` from 0.1s to 0.2s, but doesn't change.

Is see according to it's [vl6180x datasheet](https://www.st.com/resource/en/datasheet/vl6180x.pdf) it should indeed only support at least 10cm, and above that the conditions need to be good:

[image]

The [vl6180 datasheet](https://www.st.com/resource/en/datasheet/vl6180.pdf) shows up to 62cm:

[image]

For me the AliExpress TOF050C VL6180'X' works quite reliably up to 180mm.

Sadly the AliExpress product info (marketing..) shows it should work up to 50cm:

[image]

https://github.com/esphome/feature-requests/issues/745#issuecomment-1837628249

xumixu, 2025-09-30:
Does anyone have a working version for current esphome builder? 2025.9.2 ?

Seems there were many breaking changed lately.

I have tried using external components and geting the sensor in components without luck.

NonaSuomy, 2025-10-06:
Tested on 2025.10-dev seems to still work?
There was only a word change from scaling to scaler updated [here](https://github.com/esphome/feature-requests/issues/1624#issuecomment-2799970169)

```
[05:33:33.537][D][vl6180x:249]: Distance: 192 mm
[05:33:33.539][D][sensor:131]: 'VL6180X Distance': Sending state 192.00000 mm with 0 decimals of accuracy
[05:33:38.535][D][vl6180x:249]: Distance: 185 mm
```

mikulik86, 2026-10-08:
Wow! After a year still nothing...

mikulik86, 2026-10-08:
Following up on my comment above: I ran into the problems k-korn described earlier in this thread and fixed them in my own copy of NonaSuomy's component (from the `vl6180x-pololu-scaler` branch): https://github.com/mikulik86/esphome-vl6180x

Changes compared to the original:

- **Measurements don't block.** `update()` starts a measurement and returns; the result is collected with ESPHome's scheduler (first check after 10 ms, then every 5 ms, 500 ms timeout as before). With nothing in range, the old code held the whole main loop for about 50–60 ms per sensor and update, which caused the "took a long time for an operation" warnings. The ALS measurement uses the same approach, but I haven't tested ALS.
- **No warning when nothing is in range.** Range status codes 6–16 (no target, weak signal, target too close or too far, too much ambient light, wrap around filter) still report the distance as unknown, but no longer set the component's warning flag, which logged "set Warning flag" and made the status LED blink. Real faults (sensor system errors, I2C failures, timeouts) still set it.
- **Sensor name in the log lines**, so several sensors can be told apart, plus a `distance: unknown` line at debug level when nothing is in range.
- **`log_distance: false`** option to hide one sensor's distance lines.

```yaml
external_components:
  - source: github://mikulik86/esphome-vl6180x@main
    components: [ vl6180x ]
```

All other options are the same as in NonaSuomy's version. The code changes were written by Claude, Anthropic's AI assistant; I tested them with ESPHome 2026.9.1 (ESP-IDF) on an ESP32-S2 with two VL6180X behind a TCA9548A. @NonaSuomy, feel free to take any of it into your branch.

NonaSuomy, 2026-10-09:
Send me a PR or tell Claude to do it 😆

NonaSuomy, 2026-11-13:
Merged

NonaSuomy, 2026-11-14:
For any one still following along please test

```yaml
esphome:
  name: vl6180x-test

esp32:
  variant: esp32s3
  framework:
    type: esp-idf

external_components:
  - source: github://pr#22689
    components: [vl6180x]

logger:

i2c:
  id: sensor_bus
  sda: GPIO19
  scl: GPIO18
  scan: true

vl6180x:
  - id: hand_sensor
    i2c_id: sensor_bus
    address: 0x29
    update_interval: 500ms
    enable_pin:
      number: GPIO23
      mode: OUTPUT_OPEN_DRAIN
    interrupt_pin: GPIO25
    recovery:
      failure_threshold: 3
      retry_interval: 2s
      max_retries: 60

sensor:
  - platform: vl6180x
    vl6180x_id: hand_sensor
    range:
      name: "Distance"
      scaler: 3
    lux:
      name: "Illuminance"
      gain: 2.5
      update_interval: 5s

text_sensor:
  - platform: vl6180x
    vl6180x_id: hand_sensor
    range_status:
      name: "Sensor Status"
```