esphome/feature-requests #1626: Support for Wemos S2-Mini (ESP32-S2FN4R2)

dvisser, 2022-02-05:
**Describe the problem you have/What new integration you would like**

I wish to use the Wemos S2-mini with esphome as it is a very small module (similar to the Wemos D1 mini with more pins)

**Please describe your use case for this integration and alternatives you've tried:**

I have tried all possible variants currently available for the ESP32-S2 and I keep getting the same error "Configuration does not match the platform of the connected device. Expected an ESP32 device."

**Additional context**

I assume this is because there isnt a platformio board type in existence for this S2-Mini yet? I have bumped a feature request (issue) in platformio github requesting S2-Mini support also.

dvisser, 2022-02-07:
I have submitted a PR for the S2 Mini board on platformio/platform-espressif32 here https://github.com/platformio/platform-espressif32/pull/722

mories76, 2022-02-22:
@dvisser Doe you know of a way to get up and running with these boards ?
Is it possible, for example, to copy the board json from your PR in the ESPhome container ?

dvisser, 2022-02-23:
@mories76 I am unsure and I have been unsuccessful thus far in getting the S2 to work at all with ESPHome. It keeps telling me it isnt recognised as an ESP32-S2 device. 

I have resorted to using arduino and custom code in the short term until some savvy ESPHome dev / platformio person comes along and helps sort this. 

I am surprised more people arent wanting to use the Wemos S2 mini as its a tiny powerful board.

pklitscher, 2022-03-03:
Plus one for this. If I remotely knew how to get it working I'd help with the PR. I got some Wemos S2s and I love their form factor. Fingers crossed someone with the smarts picks it up

jhamhader, 2022-03-05:
Managed to build and flash successfully.

See https://github.com/platformio/platform-espressif32/pull/736

dvisser, 2022-03-05:
@jhamhader excellent work thank you!

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/996

closed 2022-11-21 by jesserockz