esphome/feature-requests #1627: Add midea_ir.follow_me action

rmounce, 2022-02-05:
**Describe the problem you have/What new integration you would like**
There is no `midea_ir.follow_me` action equivalent to `midea_ac.follow_me`. The latter works with `midea_ir` devices, however it is only available after configuring the wired `midea` component.

**Please describe your use case for this integration and alternatives you've tried:**

My AC indoor unit's return air temp sensor is influenced by the ambient temperature in the roof, basically the outside temperature. Without adjusting the setpoint the actual indoor temperature becomes too cool during the day & too warm overnight. The wall controller supports "follow me" however this room is unoccupied overnight. The solution of course is IR "follow me" which thankfully is supported by this ducted system, so a more accurate temperature reading can be provided (and in my case, combines data from multiple sensors based on which zones are enabled).

As a workaround I have configured `midea` with a dummy UART and it is functioning, however this results in logspam as it cannot detect any physically attached AC.

Workaround config:
```
remote_transmitter:
  pin: 14
  carrier_duty_percent: 50%
  id: ir_tx

# dummy
uart:
  baud_rate: 9600
  rx_pin: 16 # unpopulated
  tx_pin: 17 # unpopulated

climate:
  - platform: midea
    id: midea_uart_dummy
    transmitter_id: ir_tx
  - platform: midea_ir
    id: midea_ac
    name: "AC"
    sensor: frontend_temp

sensor:
  - platform: homeassistant
    id: frontend_temp
    entity_id: sensor.ble_weighted_avg_temp   # Sensor from HASS
    internal: true
    filters:
      - throttle: 10s
      - heartbeat: 2min             # Maximum interval between updates.
      - debounce: 1s
    on_value:
      midea_ac.follow_me:
        temperature: !lambda "return x;"
        beeper: false               # Optional. Beep on update.
```

**Additional context**

Some extra background as I have not found anyone else attempting DIY control of Midea ducted systems, and hope they find this mostly successful report. I have an ActronAir UltraSlim ducted AC with LRE-100AS indoor unit. This is a rebranded Midea system, Carrier and Rinnai also appear to sell closely related systems. In addition to the WC-02 wall controller inside the house (connected to CN40), there is a basic control panel (connected to CN10) mounted to the indoor unit in the roof with an IR window. 

I am using the new `midea_ir` component to control the system via an ESP32 placed next to the indoor unit in the roof. This works great, I am able to change the mode, temperature setpoint and fan speed via ESPHome & Home Assistant. Thanks @dudanov! This unit does not appear to have an IR Tx LED to report back its status if changes are made from the wall controller.

kryptonitecb3, 2022-04-17:
I was searching for the same thing when I came across this. My use case is similiar, I have a Danby window unit that I found out uses the Midea IR commands. The unit is 'dumb' so I cannot use UART/USB commands, and there is no feedback that I am aware of. The remote has the follow me function, and the remote commands are recognized by the remote_receiver: midea component.

It would be great if the ESPhome team could add this functionality.

kevinmilner, 2022-12-01:
Hi @rmounce, thanks for opening this issue. It looks like it may have just been fixed (?) in the next release by https://github.com/esphome/esphome/pull/4053 which is great (or maybe just progress toward a fix?).

I found this ticket by searching for "CN10" and think we may be in a similar boat, although I'm still in the research phase and wondering if I can ask you some questions before tackling this over winter break. I have some older ducted midea systems (2016) that predate the wifi dongle and have a wired controller connected (via 5 core cable labeled ABCDE) to an internal display board (single button, 3 LEDs, IR window, temperature display, located inside the housing for the ducted unit next to the circuit board), which then connects to the main circuit board via CN10. I think this is a similar system to what you have, except maybe your wired controller is connected directly to the circuit board (via CN40) rather than daisy chained to the display board in my system.

I understand that you are able to control the system via IR, and that when you make changes via the wired controller they are not broadcast via IR (so you cannot see via ESPHome and HA that a change was made), is that correct? My main question to you is the other way around: if you issue a command via IR, will the result of that command be shown on the wired remote? Lets say the system is off, and you send an IR command to turn it on, will the state of the wired remote change to reflect this? Same question for setpoint and other changes.

Ultimately, it seems that the best way to do this would be to replace the wired remote with a smarter device that can two way communicate with the system and be automated, but I'm not aware of anyone trying that route.

rmounce, 2022-12-01:
Hey @kevinmilner 
That PR looks like it solves my issue. My workaround... works so I'll wait for the next stable ESPHome release, test & report back.

Changes made via IR from Home Assistant / ESPHome are immediately reflected on my wired wall controller - mode, setpoint, fan speed.

It sounds like your system is very similar to mine, in my research I did see the wall controller variants that daisy-chain via the display board rather than connecting directly to the main board. Rinnai Australia sell a WiFi kit that daisy chains between the CN40 wall controller & internally embeds the same "USB" dongle used by their newer wall splits. So it may be possible to buy one of these kits (sold under multiple brands), replace the dongle & use the `midea` ESPHome component as-is.

https://www.ozaironline.com.au/products/rinnai-smartport-wifi-kit-lcacwifikit-for-ducted-systems
https://github.com/dudanov/iot-uni-dongle

I have also seen some discussion and some reverse engineering of the "XYE" interafce, I think this was based on Modbus / RS485? I was planning to explore these routes but quickly gave up once I discovered the one way IR option which is good enough for me. Instead my effort has been focused on controlling the zone dampers in my system :)

kevinmilner, 2022-12-02:
Awesome, thanks for the info @rmounce! I'm going to give it a try with IR then. Did you find a convenient and reliable power source for the ESP32 to tap into on the circuit board or are you just running power in externally?

rmounce, 2022-12-02:
@kevinmilner there was a standard Australian 240V outlet looping out of my indoor unit, from which the zone controller was powered. I just added a double adaptor and used an old microUSB phone charger to power the ESP32.

The wall controllers (both the CN10 and CN40 variants) run on 5 or 12VDC so there should already be some DC to tap into.
Otherwise there will be permanent 120 or 240V AC inside the unit.

rmounce, 2022-12-27:
https://github.com/esphome/esphome/pull/4053 works for me and satisfies this feature request.

Some more scenarios that should perhaps be added to <https://esphome.io/components/remote_transmitter.html?highlight=midea+transmit#remote-transmitter-transmit-midea-action>:
```
// Celsius without beep
return {0xA4, 0x82, 0x48, 0x7F, (uint8_t)(id(temp_sensor).state + 1)};
// Celsius with beep
return {0xA4, 0x82, 0x48, 0xFF, (uint8_t)(id(temp_sensor).state + 1)};
// Fahrenheit without beep (assuming temp_sensor is already in Fahrenheit)
return {0xA4, 0x82, 0x68, 0x7F, (uint8_t)(id(temp_sensor).state - 31)};
// Fahrenheit with beep
return {0xA4, 0x82, 0x68, 0xFF, (uint8_t)(id(temp_sensor).state - 31)};
```

These magic numbers seem to set
- mode: auto
- fan: auto
- setpoint 25C or 70F

But these fields seem to be ignored by the system when processing a follow me update, so it's okay to hard code them. Conversely, the temperature unit field is important and the follow me update is ignored altogether if this doesn't match the system's expectations.

For more IR format info:
- https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Midea.h
- https://docs.google.com/spreadsheets/d/1TZh4jWrx4h9zzpYUI9aYXMl1fYOiqu-xVuOOMqagxrs/edit 

I've spent some time exploring whether my Midea system can be controlled more precisely in Fahrenheit mode as a 1 degree step there is equivalent to ~0.5°C, and wondered whether is it just constrained by the IR protocol?

Short answer: not for my use case, where I have another controller that tries to [trick the Midea]<https://www.youtube.com/watch?v=P4_6e5IaQXM> into modulating its compressor speed via follow me. Everything seems to map to Celsius internally so this doesn't expose any new behaviour.

Temperature relative to setpoint
| Fahrenheit | Celsius | Observed cooling mode behaviour |
|-|-|-|
| -3 | -2 | Stop compressor
| -2 | -1 | Slow compressor to min speed
| -1 | 0 | Decrement compressor speed every 3 mins
| 0 | 0 | Decrement compressor speed every 3 mins
| 1 | 0 | Decrement compressor speed every 3 mins
| 2 | 1 | Stable compressor speed if temp dropped from setpoint+2°C/3°F. Continue decrementing compressor speed if temp increased from setpoint+1°F/0°C.
| 3 | 2 | Start compressor if off, increment compressor speed every 3 mins

closed 2022-12-27 by rmounce

flyoffacliff, 2023-01-17:
> 

As far as I am aware the wireless remote is the only one-way component. If you update the temperature on the remote (or with Wi-Fi), then the wired wall controller will update, but not the other way around.
There are many different wired wall controller models because Midea likes to overcomplicate things. Even my dealer complains about it.
There are two lines of wall controllers. The ugly old looking ones that plug into the five pin ABCDE connector, as well as the white ones that plug into CN40. If you have a white Wi-Fi orb, then you connect the wall control to that. They get daisy chained because there is only one CN40 on the board and both devices need it. I would have to dig around to find the model numbers.
I too am trying to come up with a way of setting the follow me temp without the official hardware. Probably some kind of IR bridge. They should just make the wall controls included with the system and be required. The temperature sensor is built into many of these units are a joke.

kevinmilner, 2023-01-17:
> If you update the temperature on the remote (or with Wi-Fi), then the wired wall controller will update, but not the other way around.

Unfortunately, this isn't the case for me, @flyoffacliff, even though it works for @rmounce. My wired ABCDE connector generally doesn't update when I send IR commands; it updates on/off status via the LED, but the display does not update to reflect mode or set temperature changes. It could be that I just have a different version (mine is pre-USB so I can't use the WiFi dongle approach), or the configuration of my wired connector being on CN10 (daisy changed after the IR receiver module).

Despite the wired controller not showing updates, almost everything does seem to be working well (follow me commands, on/off/mode, set changes). I can't seem to activate fan only mode, but that's not a huge deal.

I found @rmounce's table of observed behavior as a function of temperature delta to be extremely useful in my automations, thank you!

flyoffacliff, 2023-01-17:
> Unfortunately, this isn't the case for me, @flyoffacliff, even though it works for @rmounce. My wired ABCDE connector generally doesn't update when I send IR commands; it updates on/off status via [...]

What type of units? The ceiling cassette and ducted systems never had USB. You have to use a Wi-Fi smart port plugged into CN40 to get a USB port for the dongle.

Have you tried the cn40 wall control to see if that one updates properly with your system?

I did not find the table useful for for me unfortunately. The units seem to have a mind of their own sometimes and behave differently at the same temperatures. It might be because I have a multi head system with four indoor units instead of just one. Also I don't understand some of the descriptions in the table and I think it's only for cooling and not heating?

rmounce, 2023-01-19:
@kevinmilner glad table that was of value to you. I'm interested in the automations you've come up with.

I've spent far too long trying to reason about the control strategy of my ducted system. That table is a summary only for cooling mode. Behaviour is different in heating mode, and also for some period after power on?
I have a complicated script that tries to account for these cases, while also controlling my zone dampers: https://github.com/rmounce/actrl/blob/master/actrl.py

@flyoffacliff I'd expect a multi-head system to have a different, more complex control strategy :(
I'd still hope that you can get reasonable control via follow-me in each room?

flyoffacliff, 2023-01-19:
> @kevinmilner glad table that was of value to you. I'm interested in the automations you've come up with.
>
> I've spent far too long trying to reason about the control strategy of my ducted system. [...]

Nice, thanks for sharing! That must have taken a long time. I wish I could do something like that for my system.
Yes, follow-me works fine on most of my units, but I'd like to move away from using the Midea remotes for various reasons.

kevinmilner, 2023-01-19:
@rmounce

Wow, that's quite the automation, I'm impressed. So far I'm just using basic hassio yaml automations, but if I decide to go deeper down the rabbit hole, your setup has a lot of great stuff to draw from.

My setup is a bit weird in that I have a conventional (A/C, gas heat) HVAC system that I want two minis to follow. It's one big L-shaped great room of sorts, with the conventional system on one end, a mini in the middle, and another at the other end. Gas is (usually) cheaper here than electricity, so I want the bulk of the heating to come from the gas unit, and it can change the temperature for the whole space much more quickly than the minis, but I want the minis to help even things out and support it. If I set the minis to the actual target temperature, they do too good of a job holding it that the conventional system never kicks in (after it reaches temperature in the first place). That might be ideal for cooling (?), but for heating I want to use a mix. So far, I'm liking setting the minis to hold temperature at 1 degree below the target setting, which seems to result in balanced usage from both systems. I'll probably try a couple different strategies for cooling when it gets warmer and compare electricity usage.

@flyoffacliff 

> Have you tried the cn40 wall control to see if that one updates properly with your system?

I had that idea as well, but I didn't actually see it on the circuit board. I don't have a good view though, as it's up in the ceiling through a rather small access panel, so I might have missed it. The schematic sticker does list it so I may try again some day.

> What type of units? The ceiling cassette and ducted systems never had USB. You have to use a Wi-Fi smart port plugged into CN40 to get a USB port for the dongle.

Mine are some 2016 ducted systems that Midea doesn't even seem to acknowledge the existence of: B-VMH09DU-1, B-VMH12DU-1, B-VMH18DU-1, and B-VMH24DU-1. Like you, I have a multi-head system (3 on 1, and another in a separate building on it's own outdoor unit). I haven't yet figured out the intricacies of how the system responds when different units want different things, that will come another day. I don't think mine support the WiFi smart port.

zzarbi, 2023-03-04:
While the work around by using the `remote_transmitter` works, it would be nice if this was added to the `midea_ir` component. I can see some of it was implemented https://github.com/esphome/esphome/blame/dev/esphome/components/midea_ir/midea_data.h#L59

On a side note, I've also been super confused on how those HVAC system works. I've spent over 2 years trying to makes sense of it. Even the follow me with the remote or the wired controller works for about 1 hour and then either the temperature is never correct. But i can't tell if it's the remote or my units. Which is why I want to give this follow me a try.

Like other I tried the `modbus` but gave up, the wired controller `ABCDE` has 2 wire for power, one wire tells you if the AC is on/off and the last 2 is used to send the exact same signal that we do with remote but over wire (one is the inverse signal). There is no communication whatsoever from the unit to those wire controller. 

At this point I just have a basic automation, based on a bluetooth temp sensor, that turns on/off the unit with a "max" temperature (else it doesn't start at all). However I'm afraid this will kill the HVAC over time. But even that is not great because there are too many points of failure (the sensor, the bleproxy, home assistant, the IR controller) I have to have additional automation to make sure it works and have alerts so it's getting complicated.

flyoffacliff, 2023-03-05:
> @rmounce
> My setup is a bit weird in that I have a conventional (A/C, gas heat) HVAC system that I want two minis to follow. It's one big L-shaped great room of sorts, with the conventional system on one end, a mini in the middle, and another at the other end. 

I used to have a similar setup and had something hacked together with Google Assistant actually.

> @rmounce
> I haven't yet figured out the intricacies of how the system responds when different units want different things, that will come another day. I don't think mine support the WiFi smart port.

If there's a mode conflict on my systems with different units calling for different things, they will just shut down and show an error code or flashing light (depending on the model). There is an engineer mode setting that determines whether heat or cool will take priority if there's a mode conflict. I think the default is heat. Even “fan” can cause a mode conflict, and I'm not 100% sure how that factors in.

> While the work around by using the `remote_transmitter` works, it would be nice if this was added to the `midea_ir` component. I can see some of it was implemented [...]

Not sure why it would stop working after an hour, I've never had anything like that happen. Maybe you could just toggle “follow me” every hour programmatically? I do find that the newer longer Midea remotes work better than the old short ones, especially when the batteries are low.

Why did you give up on the Modbus? Was there a specific issue you ran in to?

That's interesting about the wired controller. Thanks for sharing because I haven't seen that documented anywhere. Was there a resource you learned this from? So basically the wall control knows if the system is on or off, but there is no other communication from the unit to the wall control. Not sure why it would even need to know if the system was on or off.

Yeah I actually considered doing something like that with a Zigbee temperature sensor in each room, but I was also worried about the complexity and points of failure for something as important as heating. Cycling the system on and off like that will kill the energy efficiency, because these systems are designed to run continuously and modulate the compressor and fans.
Another way to tell the unit the room temperature besides hijacking the “follow me” signals is hijacking the temperature sensor itself. It's just a 10k thermistor, and it's usually easy to access to move or remove inside the unit. Theoretically you could use an Arduino to generate a fake temperature signal if you don't want to connect an actual sensor.

flyoffacliff, 2023-03-05:
@zzarbi

zzarbi, 2023-03-05:
> @flyoffacliff  
> Not sure why it would stop working after an hour, I've never had anything like that happen. Maybe you could just toggle “follow me” every hour programmatically? I do find that the newer longer Midea remotes work better than the old short ones, especially when the batteries are low.

Honestly I have no clue, I tried it again last night and in the middle of the night the heater would stay on even thought the temperature on the remote was way higher than what I set it to. I do have the longer newer remote and this was with one of the "newest" wall unit.

> Why did you give up on the Modbus? Was there a specific issue you ran in to?

Too many unknown, I couldn't even connect, the port is in the cassette unit just (so impractical) and then I found the IR was "quicker" to a usable solution. I did found this repo though https://codeberg.org/xye/xye if anybody want to give it a try (couldn't get tit to compile either). I know there is a way to change "settings" in the program like insulation variance but neither the seller or the guy who install the unit knew how to do that.

> That's interesting about the wired controller. Thanks for sharing because I haven't seen that documented anywhere. Was there a resource you learned this from? So basically the wall control knows if [...]

I just bought a one of those cheap logic Analyzer and did it myself, I think I might have the mapping somewhere. My plan was to use those cable and build a new wall unit that would be self sufficient but I stopped progress once I had the IR automation "working". I'm using Sensibo unit flashed with `epshome`.

> Yeah I actually considered doing something like that with a Zigbee temperature sensor in each room, but I was also worried about the complexity and points of failure for something as important as [...]

That's an interesting approach. I can tell you my electric bill is not loving me and I can tell the AC is 50% of it in an average day. If I were to automate each room separately, it could mean the compressor working on max all the time so my current implementation need to be improved upon. I need some time to review your script to see if I can make use of it.

zzarbi, 2023-03-24:
FYI not sure if it's related to why `Follow me` is not working for me, but it looks like `Follow Me` on my remotes has 3 commands: (for Fahrenheit)

```
- On: 0xA4, 0x82, 0x68, 0xFF, (uint8_t)(id(${devicecode}_dht_temp).state + 1)
- Off: 0xA4, 0x82, 0x68, 0x3F, (uint8_t)(id(${devicecode}_dht_temp).state + 1)
- Update: 0xA4, 0x82, 0x68, 0x7F, (uint8_t)(id(${devicecode}_dht_temp).state + 1)
```

Now according to @rmounce, the `On` command is the same as the update command without the beep but with my newer head unit on the remote I have a "Led/Silent" button which disable beeps altogether and the display, and when this mode is "on" the remote still sends the `On` command (with the beep) but the unit doesn't make the beep because of the silent mode.