esphome/feature-requests #1628: Dreamscreen Bluetooth Control

FragMenthor, 2022-02-06:
Is is possible to add support for controlling a DreamScreen bluetooth device, on a ESP32 with BLE?
It's an Ambivision clone, and does not have WiFi (first version), but I got a Command Reference for the device. I don't know if it's possible to do this, so I leave the request and the command sheet.
THANKS to all developers! :)

[DS BLE Commands V1.pdf](https://github.com/esphome/feature-requests/files/8010786/DS.BLE.Commands.V1.pdf)

nagyrobi, 2022-02-07:
What device is that?

FragMenthor, 2022-07-04:
> What device is that?

It's an Ambilight clone, but only controlled by Bluetooth, and does not pair with the phone. It has an app, and the app, without pairing, recognizes the device and sends comanda, I don't know how, and I don't know how to help more. Maine I can scan something with nRF Connect if someone leads me the right way to do it...

alexisiglesias1, 2022-11-06:
I wish they can make this work