esphome/feature-requests #1633: Remote Transmitter: request to support "SYMPHONY" protocol

duchenpaul, 2022-02-12:
**Describe the problem you have/What new integration you would like**

One of my IR remote controller is using `SYMPHONY` (related [lib](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md)) as encode protocol which is not supported in esphome.

**Please describe your use case for this integration and alternatives you've tried:**

I tried to replay raw data, but not working.

**Additional context**

Add support for this protocol just like what we are doing [here](https://esphome.io/components/remote_transmitter.html#remote-transmitter-transmit-nec-action).

naamah75, 2022-05-17:
bump! I would like to get the `symphony `protocol too

Jilas, 2022-05-24:
Agree this would be useful, as it could well explain issues with Westinghouse Ceiling Fans (Remote ID6 - which is mentioned within the Symphony listings) as no matter how I try, I cannot send the raw signals out the fan is ignoring it.

alexberry, 2022-07-30:
Yes please, I'm also trying to get a westinghouse Model 6/ID 6 working and cannot decode with receiver, this would help me a lot.

Andyw222, 2022-07-30:
Yes please also having the same issue with the Westinghouse 6/ID6 and can't decode with the reciever

alexberry, 2022-08-02:
For anyone who wants a workaround in the interim, here's a [gist](https://gist.github.com/alexberry/56ebc81a625bb9f22964d34b45b01360) for a Westinghouse 6 Infrared send with HA integration using Symphony from [crankyoldgit/IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266)

Tazzios, 2022-10-15:
Now I know why I could not get it to work!
Here also a Westinghouse 6/ID6  user.

CharlyAutomatico, 2023-08-15:
Another one

Shadowtrance, 2023-09-02:
And another one that needs this protocol for my fan. (not Westinghouse)
Tried throwing every other type at it like raw and pronto and it won't do a damn thing. 🤣
IRremoteESP8266 works fine with it on a dev board but I have a tuya IR blaster flashed with esphome I want to use.

saltech1, 2024-04-18:
Did anyone find workaround for this issue? Even raw doesn't work for me.

Tazzios, 2024-06-13:
I cheated with replacing the westinghouse controller with the Sonoff iFan 04 with esphome flashed on it. Only the dimmer function is missing.

Jilas, 2024-06-13:
> I cheated with replacing the westinghouse controller with the Sonoff iFan 04 with esphome flashed on it. Only the dimmer function is missing.

I'd been considering the same, was the wiring as straightforward as it appears and did it fit well where the old controller was?  I assume you can turn on/off the light.  Can lose the dimming function as so rarely use that.

Tazzios, 2024-06-13:
The controller was the same size so that was easy. Light on and off is working an other limitation could be the 3 fan speeds, if you fans has more.

The wiring was a bit confusing because one stated "com" but the image on this blog made it clear it was a "N"
https://hagensieker.com/2021/09/29/sonoff-ifan04-im-a-fan/

To connect programmer i just put pins through the holes without soldering. I flashed ESPhome from the browser 
The ESPhome config that I used is this one: https://github.com/rh1rich/esphome-ifan04

Jilas, 2024-06-13:
Thanks for confirming all of that @Tazzios  My fans are all 3 speeds, so sounds like I need to get one with the next fan and start the madness of swapping them out.  Glad it can be flashed with esphome. 

Appreciate the info.

dwienie, 2024-06-29:
I am also looking for a solution to have the symphony IR codes in Esphome.

oyooyo, 2024-08-20:
I don't understand what the problem with this protocol is, does somebody understand and can explain?

I have a Westinghouse ceiling fan that seems to use this protocol. For the last couple of years I had an ESP8266-based IR-blaster with Tasmota firmware and I was able to control the ceiling fan using raw encoding without any issues.

Now I'm trying to replace that with a new Beken BK7231N-based IR-Blaster running esphome firmware.

So I'm sending the same raw signal that worked perfectly fine with Tasmota firmware - but the fan does not react.

Switching from the old IR-Blaster with Tasmota to the new one with Esphome worked fine for another device that I control via IR with raw encoding, so I guess my esphome firmware is configured correctly for transmitting IR.

Jilas, 2024-08-20:
Nice work and very interesting results.

It's been a while, so I'll have to dig out my old setup, but try this.

I'll post an update once I have.

> I think I've figured out the root of my problem. I just hooked my logic analyzer to one of the IR leds and captured the signal that the IR led actually sends when I send one of the SYMPHONY codes to control my ceiling fan.
> [image]
> 
> I sent this signal via the `remote_transmitter.transmit_raw` action, with 8 repetitions. The pause between each repetition should be about 8ms - but as you can see, **the pause actually becomes longer and longer with each repetition**, after the 7th repetition it is about 174ms instead of 8ms.
> 
> When I send the same raw signal but manually repeat the signal 4 times (instead of specifying repeat: times: 4), the signal is sent correctly:
> [image]
> ...and my ceiling fan properly reacts to the command.
> 
> So it seems to me that there is a bug in the repeat signal logic of the `remote_transmitter.transmit_raw` action, at least when the device has a Beken BK7231N.

oyooyo, 2024-08-20:
> It's been a while, so I'll have to dig out my old setup, but will try this.
> 
> I'll post an update once I have.

I've meanwhile looked further into this and filed an issue. Apparently there is indeed a bug in file `remote_transmitter_libretiny.cpp` that causes the strange behaviour described above. As I understand, the problem should occur on all platforms using libretiny = all non ESP8266/ESP32 platforms.

And the problem should occur with any IR protocol when using multiple repetitions, not just with raw encoding. I assume that for most other IR protocols those too-long pauses are no big problem (as I could for example control a device that uses the RC5 protocol just fine), but they seem to be a major problem with that "SYMPHONY" protocol.

By the way, the library linked in the OP is using 400µs and 1250µS timings for this "SYMPHONY" protocol. While that probably works fine in most cases, somebody implementing this protocol in esphome should probably be using 422µs / 1266µs instead, as these seem to be the ideal values according to my understanding of the datasheet:
https://www.alldatasheet.com/datasheet-pdf/pdf/124369/ANALOGICTECH/SM5021B.html