esphome/feature-requests #1635: Allow defining and filling “user attributes” on sensors

DrCoolzic, 2022-02-13:
**Describe the problem you have/What new integration you would like**

It is possible to set the value of some predefined attributes (e.g. device_class, state_class…) of a sensor in the yaml configuration file. But I believe it is not possible to define “user attributes” and to return values for such attributes programmatically.
The goal of this request is to allow for a specific sensor to return an extra property with an associated value. 

**Please describe your use case for this integration and alternatives you've tried:**

Let’s take the example of the Teleinfo components. This component allows to retrieve data from a French electrical meter. For most of the information, we just need to be transmitting a value for example the instantaneous “energy value”. But for some tags we need to send back two information—for example the “maximum power of the day” has a timestamp associated that indicates when the maximum value happened. The only possibility to transmit this information currently would be to define two “related sensors”: one that send the power value and one that send the timestamp value but this is not a good solution. 

What would be nice is to find a way to transmit the timestamp as a property attached to the sensor.
So in this example we would have a sensor "max power" with a property "timestamp" 

**Additional context**

nagyrobi, 2022-07-19:
I'd find that useful for other integrations too, like switch etc.

DrCoolzic, 2022-07-21:
Yes I hope someone will look at it :)

nagyrobi, 2022-07-21:
It needs modifications to the API and Home Assistant component side. Could go along with https://github.com/esphome/feature-requests/issues/1568

nagyrobi, 2022-09-18:
https://github.com/esphome/feature-requests/issues/1112

juliekoubova, 2022-11-13:
I can try and look into it, https://github.com/esphome/esphome/pull/3825 could use them, too. 

It currently needs the user to define a separate binary sensor for each of the "attribute" states they are interested in, and there are only about eight different ones, so it would be a perfect fit for attributes.

jeremysherriff, 2024-03-01:
I assume we're still limited to exposing several sensors to HA and then using a template to combine things? I feel that this is an artificial limitation - the use-case of updating an icon based on state should be a good enough reason by itself, especially given [the recent changes in HA](https://www.home-assistant.io/blog/2024/02/07/release-20242/#icons-everywhere)

kquinsland, 2024-04-14:
> It needs modifications to the API and Home Assistant component side. Could go along with #1568

Now that 2024.04 has landed and there's [native support for tagging things](https://www.home-assistant.io/blog/2024/04/03/release-20244/#labels-tag-everything-any-way-you-want), there should only be a bit of work on the [mqtt disovery ](https://github.com/home-assistant/core/blob/dev/homeassistant/components/mqtt/discovery.py) side of things and adjusting the [mqtt component](https://github.com/esphome/esphome/blob/dev/esphome/components/mqtt/__init__.py#L491) in ESPHome?

asergunov, 2024-10-20:
Hi. I really interested to implement it. For me the main goal to be able to send large amount of data with state. I'd like to send something like oscillogram to my sensor reading. But state in HA is always string and it's limited by 255 chars. Attributes in other hand doesn't have such limitation. [Docs](https://www.home-assistant.io/docs/configuration/state_object/)

I dig to a problem a bit. Here what I found:

There is protobuf-based protocol to send data to HA [esphome/aioesphomeapi)](https://github.com/esphome/aioesphomeapi)
  * `SensorStateResponse` is the one who serializes the state [usage](https://github.com/esphome/esphome/blob/c019ff34bccf08b3070cae3c1e71c2767cf70a6e/esphome/components/api/api_connection.cpp#L458C6-L467)
  * `ListEntitiesSensorResponse`is separate message to transfer states. [usage](https://github.com/esphome/esphome/blob/c019ff34bccf08b3070cae3c1e71c2767cf70a6e/esphome/components/api/api_connection.cpp#L485)
  * There is sometimes `*CommandRequest` defined to serialize command

Here is [all 3 messages for Fan](https://github.com/esphome/aioesphomeapi/blob/c9a80348ce6130daeb4f753392c45cb445afe16f/aioesphomeapi/api.proto#L384-L417)

In HA database they all just part of attributes field and we have to store them all at once. I like this design so we send attributes can't change only once device connected and the send changeable every state change but with respect of data type. This helps a lot with message sizes. 

I like to extend these messages with attributes field keeping the typization using `oneof` keyword. So we could send any custom attributes on any stage of conversation.

Only thing I didn't figure out yet is where these messages become a hassio db modification commands. Could any one help me with that?

DIYtechie, 2025-05-02:
I also believe this would be very useful for a range of things. My use case is a ESPHome HDMI CEC controller. I just figured out how to expose the raw and translated HDMI CEC commands as sensors, but this traffic is not relevant to store so would be better as attributes to a ESPHome status sensor. That way you don’t need multiple sensors and yiu don’t need to exclude from logbook and recorder.

DIYtechie, 2025-06-03:
I’ve looked into how this could be implemented natively in ESPHome with minimal friction. Based on a quick exploration of the source code, it appears relatively straightforward to support sending custom attributes from ESPHome sensors to Home Assistant via the native API.

This functionality would be extremely useful for many use cases and could reduce reliance on MQTT for those needing rich sensor metadata.

Implementation Overview

The idea is to allow Sensor objects to expose a stringified JSON object representing their custom attributes, which would be included in the SensorStateResponse message sent to Home Assistant.
### 1. Modify `api.proto`

Add a new field to the `SensorStateResponse` message:

```
message SensorStateResponse {
  uint32 key = 1;
  float state = 2;
  string custom_attributes_json = 3;  // new field
}

```
This allows ESPHome to send arbitrary attribute data in a flexible, compact format.

### 2. Extend the Sensor class

In `sensor.h`:

```
class Sensor : public EntityBase {
 public:
  ...
  void set_custom_attributes_json(const std::string &json) {
    this->custom_attributes_json_ = json;
  }

  const std::string &get_custom_attributes_json() const {
    return this->custom_attributes_json_;
  }

 protected:
  std::string custom_attributes_json_;
};
```

This allows users to define attributes at runtime in their custom components or lambdas.

### 3. Modify `api_service.cpp`

In the code that constructs `SensorStateResponse`, include the new field:

```
res->set_key(sensor->get_object_id_hash());
res->set_state(state);
res->set_custom_attributes_json(sensor->get_custom_attributes_json());
```

### 4. YAML Usage Example

A possible way for users to define attributes in YAML (future idea):

```
sensor:
  - platform: template
    name: "My Sensor"
    lambda: |-
      id(my_sensor).set_custom_attributes_json("{\"room\": \"garage\", \"status\": \"ok\"}");
      return 42.0;
```

### 5. Home Assistant Changes

Home Assistant’s ESPHome integration would need a small update to consume and expose the new `custom_attributes_json` field.

Assuming this lives in the `esphome` component, changes would be needed in the Python entity class, typically found in [components/esphome/](https://github.com/home-assistant/core/tree/dev/homeassistant/components/esphome).

In the Python entity class (e.g. `EsphomeSensor`), modify or extend the `async_update_callback` to parse the field:

```
import json

def async_update_callback(self, state):
    self._state = state.state

    # Parse custom attributes
    if hasattr(state, "custom_attributes_json") and state.custom_attributes_json:
        try:
            self._attributes = json.loads(state.custom_attributes_json)
        except ValueError:
            self._attributes = {}
    else:
        self._attributes = {}
```

Then expose the attributes with:

```
@property
def extra_state_attributes(self):
    return self._attributes or {}
```

This will allow ESPHome sensors to show rich attributes in the Home Assistant UI just like MQTT sensors do today.

Notes
- This approach avoids Protobuf map<string, string> limitations and sticks to JSON for flexibility and safety.
- The Home Assistant part is minimal and does not affect existing users.
- This patch is not yet tested. The code above was generated by ChatGPT based on a brief review of the ESPHome and Home Assistant codebases and should be considered a draft proposal for a proper programmer to double check and test out.

asergunov, 2025-06-03:
Performance wise I'd prefer to extend probuf protocol to avoid converting numbers to stings on device side. Thanks for pointing the code on home assistant integration side! I'll try to find time this weekend for that.

DIYtechie, 2025-06-23:
> Performance wise I'd prefer to extend probuf protocol to avoid converting numbers to stings on device side. Thanks for pointing the code on home assistant integration side! I'll try to find time this weekend for that.

Any luck getting it to work? :)