esphome/feature-requests #1636: DCF77 component RF transmitter

tiagofreire-pt, 2022-02-13:
**Describe the problem you have/What new integration you would like**
I want to create a DCF77 clock emulator.

**Please describe your use case for this integration and alternatives you've tried:**
Sync my home clocks through a DCF77 RF signal generated by a ESP8266 or ESP32, with its time source from sntp, gps or home assistant

**Additional context**

Similar projects:

https://github.com/SensorsIot/DCF77-Transmitter-for-ESP32
https://www.elektormagazine.com/magazine/elektor-201803/41373
https://www.0x7.ch/text/dcf77.pdf

KiLLeRRaT, 2024-04-24:
Here is another similar project: https://github.com/EddyVB/ESP8266-NTP2DCF77

sst1xx, 2025-03-16:
Hello @tiagofreire-pt, please feel free to test or modify the project at https://github.com/sst1xx/radio_cron_dcf77?tab=readme-ov-file#using-with-esphome. I'm currently using it with my radio-clock HD-1688 and would appreciate any feedback or improvements you might suggest.