esphome/feature-requests #1638: Button Custom Component (Enabling platform button.custom ?)

ClemensSchartmueller, 2022-02-15:
**Describe the problem you have/What new integration you would like**

Support for custom button components

**Please describe your use case for this integration and alternatives you've tried:**

For e.g. to trigger advanced actions on an esp8266 that cannot be easily implemented via esphome YAML directly. In my case it is a custom implemented catfeeder with servo and LED+LDR as encoder. An "ugly" workaround would be to use a switch that automatically resets its state.

**Additional context**

The Button class already provides a virtual press_action method to be overwritten for custom button implementations. (Not knowing ESPHome's core logic) adding the configuration platform button.custom may already do the trick? Tutorial and doc could be easily adapted from the "custom switch component" documentation.

rum0s, 2022-07-02:
+1 That would also be helpful for me, triggering a 433 command.
At the moment I also use the "ugly" workaround...

nagyrobi, 2022-07-07:
Using a Template Button this is possible: https://esphome.io/components/button/template.html

closed 2022-07-07 by nagyrobi